PX4/PX4-Autopilot #10336: NXPhlite: sdcard not working

Nielesonimusso, 2018-08-27:
board cannot access and use SD card. Attached two logs from console what happens when sd card is inserted when the NXPhlite is already turned on, and during boot. One is for an SD card with 4gb capacity, one with an SD card with 8gb capacity. We have also tried a 16gb SD card, with the same results as the 8gb card. Actions are annotated in the log, starting with `###`

[sdcard crash (reboot 4gb).txt](https://github.com/PX4/Firmware/files/2322948/sdcard.crash.reboot.4gb.txt)

[sdcard crash (reboot 8gb).txt](https://github.com/PX4/Firmware/files/2322949/sdcard.crash.reboot.8gb.txt)

Note that when inserting the SD card while the NXPhlite is already on, it seems that the I2C bus is influenced: Mag #0 (HMC5883, the (external) compass in our uBlox M8N) gets a timeout. The same holds for Baro #0, which is the MPL3115A2, is also connected on the I2C bus.

davids5, 2018-09-11:
@Nielesonimusso 

The sd card is not expected to be removed/inserted during operation. So the main concern would be insertion followed by reset not working.

In the log I see 
`INFO  [tone_alarm] missing command, try 'start', status, 'stop'`

This was fixed when https://github.com/PX4/Firmware/pull/10257 came in on 8/30.

The timeouts are related to long threads running on the work queue.

Please retest on current master.
This is what I get, with no card
```
## NO CARD - Error Tone
Successfully bound SDHC to the MMC/SD driver
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: NXPHLITE_V3
FW git-hash: 37c12c3b5ccf56478364fdc3392b87c8c6d2f554
FW version: 1.8.0 0 (17301504)
OS: NuttX
OS version: Release 7.22.0 (118882559)
OS git-hash: 0ffe7ce8f6f78190c714e6daa866f73ce271a5bd
Build datetime: Sep 11 2018 04:45:01
Build uri: localhost
Toolchain: GNU GCC, 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
MFGUID: 0022ffffffffffff4e4536207004000f
MCU: Kinetis K66, rev. 8
UID: ffff2200:ffffffff:2036454e:0f000470
nsh: mount: mount failed: No such device
INFO  [tune_control] Publishing standard tune 16
nsh: mkfatfs: mkfatfs failed: No such device
INFO  [tune_control] Publishing standard tune 2
WARN  [dataman] Could not open data manager file /fs/microsd/dataman
ERROR [dataman] dataman start failed
INFO  [load_mon] stack check enabled
WARN  [mtd] mtd needed 2 attempts to attach
INFO  [param] selected parameter default file /fs/mtd_params
WARN  [rgbled] no RGB led on bus #2
WARN  [blinkm] I2C init failed
WARN  [blinkm] init failed
ERROR [param] Parameter SYS_USE_IO not found
WARN  [bst] no devices found
adc CDev::init
WARN  [hmc5883] no device on bus 1 (type: 2)
HMC5883_I2C on I2C bus 2 at 0x1e (bus: 100 KHz, max: 400 KHz)
WARN  [mpl3115a2] no device on bus 2
MPL3115A2_I2C on I2C bus 2 at 0x60 (bus: 100 KHz, max: 400 KHz)
FXOS8701CQ on SPI bus 2 at 0 (1000 KHz)
INFO  [fxos8701cq] accel cutoff set to 30.00 Hz
FXAS21002C on SPI bus 2 at 1 (2000 KHz)
INFO  [fxas21002c] gyro cutoff set to 80.00 Hz
ERROR [sensors] FAILED APPLYING mag  CAL #2
ERROR [sensors] FAILED APPLYING mag  CAL #2
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
ERROR [mavlink] DM_KEY_MISSION_STATE lock failed
ERROR [mavlink] offboard mission init failed (-1)
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS4 @ 57600B
INFO  [init] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0
INFO  [logger] logger started (mode=all)
INFO  [logger] log root dir created: /fs/microsd/log
## NO CARD - Error Tone

## Card insertion - Error Tone stuck on
NuttShell (NSH)
nsh> ERROR [sensors] Mag #0 fail:  TIMEOUT!
ERROR [sensors] Sensor Mag #0 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 0: Mag #0 priority: 1
WARN  [sensors] Remaining sensors after failover event 0: Mag #1 priority: 0
WARN  [ecl/EKF] EKF stopping navigation
INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)

## Reset - normal operation
nsh> Successfully bound SDHC to the MMC/SD driver
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: NXPHLITE_V3
FW git-hash: 37c12c3b5ccf56478364fdc3392b87c8c6d2f554
FW version: 1.8.0 0 (17301504)
OS: NuttX
OS version: Release 7.22.0 (118882559)
OS git-hash: 0ffe7ce8f6f78190c714e6daa866f73ce271a5bd
Build datetime: Sep 11 2018 04:45:01
Build uri: localhost
Toolchain: GNU GCC, 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
MFGUID: 0022ffffffffffff4e4536207004000f
MCU: Kinetis K66, rev. 8
UID: ffff2200:ffffffff:2036454e:0f000470
nsh: hardfault_log: command not found
INFO  [tune_control] Publishing standard tune 1
INFO  [dataman] Unknown restart, data manager file '/fs/microsd/dataman' size is 350216 bytes
INFO  [load_mon] stack check enabled
WARN  [mtd] mtd needed 2 attempts to attach
INFO  [param] selected parameter default file /fs/mtd_params
WARN  [rgbled] no RGB led on bus #2
WARN  [blinkm] I2C init failed
WARN  [blinkm] init failed
ERROR [param] Parameter SYS_USE_IO not found
WARN  [bst] no devices found
adc CDev::init
WARN  [hmc5883] no device on bus 1 (type: 2)
HMC5883_I2C on I2C bus 2 at 0x1e (bus: 100 KHz, max: 400 KHz)
WARN  [mpl3115a2] no device on bus 2
MPL3115A2_I2C on I2C bus 2 at 0x60 (bus: 100 KHz, max: 400 KHz)
FXOS8701CQ on SPI bus 2 at 0 (1000 KHz)
INFO  [fxos8701cq] accel cutoff set to 30.00 Hz
FXAS21002C on SPI bus 2 at 1 (2000 KHz)
INFO  [fxas21002c] gyro cutoff set to 80.00 Hz
INFO  [commander] Mission #2 loaded, 5 WPs, curr: 0
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS4 @ 57600B
INFO  [init] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0
INFO  [logger] logger started (mode=all)

NuttShell (NSH)
nsh>

```

davids5, 2018-09-12:
@Nielesonimusso - Ahh - I think we are back feeding from the FTDI cable. 

One test to conduct will tell us.

Remove the red wire from the FTDI. (tape it off so it does not short)

1) Remove the DCD cable.
2) Power the FMU from the power module OR USB. 
3)  After the FMU is powered attach the DCD

Then run the test you did above.

Nielesonimusso, 2018-09-12:
@davids5 followed your steps, powering through USB, and the SD card works! Looks like the issue is solved this way.

closed 2018-09-12 by Nielesonimusso

davids5, 2018-09-13:
@Nielesonimusso - That is Awesome news! Thank you! 

@igalloway - please put out a publication to cut the FTDI VCC pin on the circuit board (remove the pin) ALL current version of the DCD-LZ-R1.

igalloway, 2020-04-30:
For the record, this adjustment was done on the DCD-LZ DCD-LZ-ADAPT adapter board to prevent the FTDI USB-UART cable from providing power.