PX4/PX4-Autopilot #10338: Factory mode

Stifael, 2018-08-27:
It would be very useful to have a factory mode in which vehicle specific things can be tested such as:
- landing gears
- LEDs
- motors
- ...

Stifael, 2018-08-27:
> Do you see this as something driven by the GCS?

Not necessarily, but can be extended such that certain tests can be started over GCS.

I was wondering if it would make sense to have some sort of a factor-mode, in which different tests can be executed. Motor-tests would be part of that factory-mode

DonLakeFlyer, 2018-08-27:
I don't see motor testing as only a factory mode thing. It's a basic DIY user need.

Stifael, 2018-08-27:
To me it does not matter how it is implemented at the end. With "factory-mode", I am talking more about a firmware mode in which only certain tests can be executed. I don't really care if is is an actual mode or not. In addition, my request should be independent of GCS. I also want to be able to start those tests from the shell...

ryanjAA, 2018-08-27:
Do you mean a mode like auto, stab, etc? Some understanding of the use case would be good. I can see why you’d want a factory set of actions to access from a gcs or external (shell) but a mode based approach could get scary if end users end up in it by accident. 

I’m all for factory based tests, we have a big long check that every plane has to go through prior to leaving here so automating that would be great but the approach here is crucial. I’m supportive of this high level, implmentetion though will be crucial and gcs based (even a shell script I suppose) seems much more appropriate then a demo or test mode like you see on so many electronics. I Just keep thinking of end users getting there by accident. Limiting it during flight would be easy but still, what ramifications would that have by having a drop down menu called factory? Advanced topics that limit for factory based scenarios should be less easy to access than more easy.

Maybe a parameter that requires a reboot to enable the mode? Something that can not be put in place accidentally as that will inherently and eventually occur.

AlexKlimaj, 2018-09-04:
I am working on an automated test and calibration fixture to be used on a production line. This would be really useful. Pretty much initiate calibration, read out and check the calibration result, spin up individual motors and check current, check the ultrasonic distance sensor output, etc. 

For the most part except for the spinning individual motors, I think I can do everything with what exists today.

Stifael, 2018-09-05:
> Do you mean a mode like auto, stab, etc? Some understanding of the use case would be good.

As i said, I don' t really care if it is a mode or not. what is important, however, is the fact that you already mentioned: it needs to be save and cannot create conflicts with the flightstack. I don't know how to make the request more clear without jumping directly into implementation details. I want to be able to test different components of the system independently. For instance, currently it is not possible to test gears, motors, leds etc. easily. 

> we have a big long check that every plane has to go through prior to leaving here so automating that would be great but the approach here is crucial.

I think you already went too far. First we need to have the tests in place that can be executed. Combining the tests for different purposes such as preflight-checks can be done afterwards. 

> For the most part except for the spinning individual motors, I think I can do everything with what exists today.

That sounds good. Just keep in mind that it should not create conflicts with the commander. In the past that was happening with the calibration, where the vehicle was able to arm during calibration.

closed 2019-02-18 by stale[bot]