PX4/PX4-Autopilot #10351: Mag Calibration always slightly off

sverling, 2018-08-28:
When calibrating the magnetometer it first always triggers the internal calibration [here](https://github.com/PX4/Firmware/blob/3ff8cd33a1d50b06f1ba9cdcac835208c144d3a1/src/modules/commander/mag_calibration.cpp#L221). 
For the HMC5983/HMC5883 (did not look into other sensors) this causes that the scaling values at the end of this internal sensor calibration are not equal to one anymore. After this step, the sampling of the datapoints for the regular calibration starts. This datapoints are now sampled with a scale being not equal to 1. After calculating the calibration parameters the previous scaling is overwritten [here](https://github.com/PX4/Firmware/blob/3ff8cd33a1d50b06f1ba9cdcac835208c144d3a1/src/modules/commander/mag_calibration.cpp#L822-L827) without accounting for the previous scaling. 
This causes that the calibration parameters will always be slightly off.

LorenzMeier, 2018-08-28:
Thanks! It sounds like a reordering would solve it - would you want to send a PR doing that?

sverling, 2018-08-28:
Does it actually make sense to keep this internal calibration there?
From what i saw the bmm150 always returns OK.
For the ist8310 and the lis3mdl it only does a check on the values but in the end it is ignored.
For the hmc5983 where it actually influences the scaling it in any case will overwrite it with the real calibration afterwards. So I am unsure about the gain of running it.

LorenzMeier, 2018-08-28:
For the HMC5883 the original idea / design was to apply the internal calibration on every boot in order to try to deal with scale changes during transport and handling. I'm not sure that was actually clever and makes sense. What's your take on this?

sverling, 2018-08-28:
I think the idea is good. But as i understood the code it does it on every bootup, but afterwards similar as in the calibration it just overwrites the values. So i guess a fix would be to apply the regular calibration on top of the internal one (basically multiplying the scales) instead of replacing it.

The only thing I am not sure about is the temperature compensation that it might interfere. Activating the temperature compensation before we do the internal calibration on startup might minimize that effect.

dagar, 2018-08-28:
I'm now wondering if this is contributing to some of the magnetometer inconsistent warnings.

ryanjAA, 2018-08-28:
Definitely seeing the mag inconsistency warnings on our end but only from (about) 1.8 forward.

JonReacher, 2018-08-31:
Same here. Have had to disable internal mags to fly.

dagar, 2018-11-05:
I'm working on addressing during a larger cleanup effort.
https://github.com/PX4/Firmware/tree/pr-mag_driver_cleanup

ryanjAA, 2019-09-22:
I’m still seeing this with an internal mag, an external hmc5883 and external qmc5883. Using stable (192) they don’t conflict and all report back data but report back mag inconsistencies. They are quite physically separated (400mm) and nothings changed during calibration and reboot, just have to disable 1 or sometimes 2 of them to arm. Also have to change mag_prime to the one I want to use (which makes sense).

dagar, 2020-08-14:
Fixed.

closed 2020-08-14 by dagar