PX4/PX4-Autopilot #10352: SITL test intermittent failures

dagar, 2018-08-28:
The SITL mission tests are periodically failing due to anomalies detected by the ecl analysis script. FW is the most frequent, but VTOL is also failing.

https://logs.px4.io/plot_app?log=88a0abf0-78bf-430b-8867-5821e8595f76
[image]

http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests/detail/pr-vehicle_trajectory_waypoint-limit/6/pipeline

Do we need to relax the script or is this something that should be adjusted in the gazebo models?

dagar, 2018-08-28:
More examples
 - http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests/detail/master/94/pipeline/
 - http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests/detail/master/92/pipeline/
 - http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests/detail/master/88/pipeline

dagar, 2018-08-29:
- http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests/detail/PR-10179/12/pipeline

priseborough, 2018-08-29:
The high frequency accel noise appears high relative to what we see on most flight vehicles, but I was able to find a vehicle log with equivalent levels. I will raise a PR to adjust the thresholds and also improve the analysis script message to indicate which quantity has failed.

HF accel vibe from sim log

[image]

HF accel vibe from flight log

[image]

dagar, 2018-08-30:
Fixed by https://github.com/PX4/Firmware/pull/10366

closed 2018-08-30 by dagar