PX4/PX4-Autopilot #10358: Mavlink v1/v2 autonegotiate with Sik radio review

dagar, 2018-08-29:
Should mavlink (PX4 + QGC) automatically select v1 in the presence of an older Sik radio that doesn't support mavlink v2?

I've been seeing a lot of logs from setups with Sik radios that are missing RADIO_STATUS and in v2 mode, resulting in poor performance.

@bkueng @DonLakeFlyer @LorenzMeier any input?

DonLakeFlyer, 2018-08-29:
It should already do that. @LorenzMeier Implemeneted a while ago. I not fully up to date on how it works. Basics are it sends a mavlink 2 ping type of message to the radio. if the radio responds it supports 2 if not, it doesn.t. Then QGC uses that to determine what to do with the connection.

dagar, 2018-08-29:
Then it appears to not be working.

hamishwillee, 2018-08-29:
@dagar Docs for what is supposed to happen for [SiK here](https://mavlink.io/en/guide/mavlink_version.html#semi-transparent-legacy-radios)

LorenzMeier, 2018-08-29:
@dagar Did you update your SiK radios via QGC? Watch the firmware URL, it should come off the PX4 AWS bucket.

dagar, 2018-08-29:
Yes I've updated mine, but it's the large existing user base I'm worried about.

closed 2019-02-18 by stale[bot]