PX4/PX4-Autopilot #10360: Altitude performance on 1.8 sometimes worse than on 1.7

AuterionWrikeBot, 2018-08-29:
Idea of Paul:
back to back testing on same hardware. Run 1.8 estimator on 1.7 and 1.8

AuterionWrikeBot, 2018-08-29:
➤ Thomas Gubler commented:

@Roman Bapst could you please provide the logs to paul?

AuterionWrikeBot, 2018-08-29:
➤ Paul Riseborough commented:

Relates to https://github.com/PX4/Firmware/issues/9740

jfcisneros, 2018-08-30:
The following logs were obtained with a similar vehicle to the one referred in https://github.com/PX4/Firmware/issues/9740#issuecomment-413448792 . 
**Altitude hold**
v1.7.3    
https://logs.px4.io/plot_app?log=f8cba56b-6eda-4ec0-8722-6dabc823f4c8
v1.8.0
https://logs.px4.io/plot_app?log=c4fddaa4-ef0f-4afb-bcb4-f9af3fcd396a
**Position Control (No commands)**
v1.7.3
https://logs.px4.io/plot_app?log=169ba77f-b011-4906-ad21-99a231776ab6
v1.8.0
https://logs.px4.io/plot_app?log=000b2a95-ef3e-40fe-8d82-4fcec21e7121

AuterionWrikeBot, 2018-08-30:
➤ Wrike Bot commented:

@Paul Riseborough All subtasks have been completed or deferred, you might want to complete this task as well.

priseborough, 2018-08-30:
I was able to compare the v1.7.3 and v1.8.0 logs and check the EKF baro lat innovations and the consistency of the output vertical velocity and vertical position derivative. Both metrics indicated that the ekf  in the 1.8.3 log was performing  better than in the 1.7.3 log. 

v1.7.3 baro innovations
[image]

v1.7.3 baro innovations
[image]

v1.8.0 vertical velocity and vertical position derivative
[image]

v1.7.3 vertical velocity and vertical position derivative
[image]

priseborough, 2018-08-30:
@jfcisneros Plotting the baro alt and distance sensor data overlayed shows a significant inconsistency. Can you  please confirm your sensor calibration. If the baro has this much scale factor error, then it is pointless doing 1.7.3 vs 1.8.0 comparisons.

v1.7.3
[image]

v1.8.0
[image]

tops4u, 2018-08-31:
Could the Barometer be exposed to Downwash/Turbulence? I'm interested in this Issue since I have also seen rather large Oscillations of my F-450 Quad with a Pixracer, but now when I think of it, it is probably exposed to Downwash Effects (it does not have LIDAR/US Sensors)

priseborough, 2018-09-02:
That's possible. A hand held height change test without motors running will confirm if it is a sensor or aerodynamic issue.

jfcisneros, 2018-10-17:
Sorry for the delay. I just did the recommended testing without the motors running.
Both of the logs were recorded with v1.8.0
**Hand Held Test**
https://review.px4.io/plot_app?log=4c726610-db8d-472b-924f-2359e13b0f54
[image]

**Altitude Hold**
https://review.px4.io/plot_app?log=074083e5-d457-47fb-8f96-f16c36ace664
[image]

There seems to be a great effect in the baro scale factor while flying. 
For our vehicle we have selected Range sensor as EKF2_HGT_MODE, and Range aid enabled for EKF2_RNG_AID. Wouldnt this set the EKF2 to use Range Finder height rather than height from the barometer?