PX4/PX4-Autopilot #10368: Make command on Macbook Pro 2018 with macOS High Sierra 10.13.6 doesn't work

LeoMustafaNXP, 2018-08-30:
**Describe the bug**
I try to build the nxphlite Firmware on a MacBook Pro 2018 with macOS High Sierra 10.13.6.
I get the following bug.

`LeoMacBookPro:Firmware leutrimmustafa$ make nxphlite-v3_default
ninja: Entering directory /Users/leutrimmustafa/Desktop/PX4/Firmware/build/nuttx_nxphlite-v3_default
[249/874] ROMFS: copying extras
FAILED: ROMFS/romfs_extras.stamp 
cd /Users/leutrimmustafa/Desktop/PX4/Firmware/build/nuttx_nxphlite-v3_default/ROMFS && /usr/local/Cellar/cmake/3.12.1/bin/cmake -E make_directory /Users/leutrimmustafa/Desktop/PX4/Firmware/build/nuttx_nxphlite-v3_default/genromfs/extras/ && /usr/local/Cellar/cmake/3.12.1/bin/cmake -E copy_directory /Users/leutrimmustafa/Desktop/PX4/Firmware/build/nuttx_nxphlite-v3_default/romfs_extras/ /Users/leutrimmustafa/Desktop/PX4/Firmware/build/nuttx_nxphlite-v3_default/genromfs/extras/ && /usr/local/Cellar/cmake/3.12.1/bin/cmake -E touch romfs_extras.stamp
Error copying directory from "/Users/leutrimmustafa/Desktop/PX4/Firmware/build/nuttx_nxphlite-v3_default/romfs_extras/" to "/Users/leutrimmustafa/Desktop/PX4/Firmware/build/nuttx_nxphlite-v3_default/genromfs/extras/".
[262/874] Generating nuttx/drivers/libdrivers.a
ninja: build stopped: subcommand failed.
make: *** [nxphlite-v3_default] Error 1
LeoMacBookPro:Firmware leutrimmustafa$ `

**Software versions**
- cmake version 3.12.1
- GNU Make 3.81
- PX4 Firmware from the master branch 
- MacBook Pro 2018 (15-inch) with macOS High Sierra 10.13.6

thank you for your support.

LorenzMeier, 2018-08-30:
@davids5

tops4u, 2018-08-31:
I just tested this (slightly different HW though). 

- MacBookPro 13" 2017 (Modell 14,1)
- cmake version 3.10.0
- GNU Make 3.81
- Latest Master from today

Make Output: 
```
-- PX4 VERSION: v1.8.0-1061-g924611016
-- CONFIG: nuttx_nxphlite-v3_default
-- Build Type: MinSizeRel
(...)
-- The ASM compiler identification is GNU
-- Found assembler: /usr/local/bin/arm-none-eabi-gcc
-- Found PythonInterp: /usr/bin/python (found version "2.7.10") 
-- Found PY_jinja2: /Library/Python/2.7/site-packages/jinja2  
-- C compiler: arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
-- C++ compiler: arm-none-eabi-g++ (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
-- PX4 ECL: Very lightweight Estimation & Control Library v0.9.0-583-g56b8bb0
-- Using C++11
(....)
```

Works like a charm.

reddigfabian, 2018-08-31:
Having the same exact issue as @LeoMustafaNXP with same exact specs/versions.

fredowski, 2018-09-02:
Hi, I have the same problem when running `make px4fmu-v4_default` with this setup
  * MacOS 10.13.6
  * cmake 3.12.1
  * make 3.81

The release notes for cmake 3.12 

https://cmake.org/cmake/help/v3.12/release/3.12.html#id1

say that `cmake copy_diretory` now fails when the source directory does not exist. The previous behavior was that the target directory was created empty.

reddigfabian, 2018-09-02:
Looks like downgrading cmake solves this issue. Not yet tested to see if the firmware is fully functional

fredowski, 2018-09-02:
I tried to build with cmake 3.11.4 and then the build works.

fredowski, 2018-09-02:
With the following patch I can build also with cmake 3.12.1:

```
diff --git a/ROMFS/CMakeLists.txt b/ROMFS/CMakeLists.txt
index 94cd9cf98..ff489d62b 100644
--- a/ROMFS/CMakeLists.txt
+++ b/ROMFS/CMakeLists.txt
@@ -105,7 +105,7 @@ endif()
 
 add_custom_command(OUTPUT romfs_extras.stamp
        COMMAND ${CMAKE_COMMAND} -E make_directory ${romfs_gen_root_dir}/extras/
-       COMMAND ${CMAKE_COMMAND} -E copy_directory ${PX4_BINARY_DIR}/romfs_extras/ ${romfs_gen_root_dir}/extras/
+       COMMAND if [ -d  ${PX4_BINARY_DIR}/romfs_extras ]; then  ${CMAKE_COMMAND} -E copy_directory ${PX4_BINARY_DIR}/romfs_extras/ ${romfs_gen_root_dir}/extras/ \; fi
        COMMAND ${CMAKE_COMMAND} -E touch romfs_extras.stamp
        DEPENDS
                ${romfs_gen_root_dir}/init.d/rcS
```

mhkabir, 2018-09-02:
@fredowski Can you please send a PR with that patch? Thanks!

PR #10402 (2018-09-04): cmake build - check directory existance before copy_directory
Starting with cmake version 3.12 the cmake -E copy_directory
command fails if the source directory does not exist. This results
in a build failure. This fix checks the existance of the
source directory before the copy_directory command ist started.

Closes: #10368

**Describe possible alternatives**
I could not find a cmake only way without the use of shell script to achieve this
commit ad6e48c3 (2018-09-04): cmake - check directory existence before copy_directory (#10402)

Starting with cmake version 3.12 the cmake -E copy_directory
command fails if the source directory does not exist. This results
in a build failure. This fix creates the source directory before
the copy which does not harm if the source directory exists.

Closes: #10368
closed 2018-09-04 by dagar