PX4/PX4-Autopilot #10376: **BUG** Timed Loiter running indefinitely

JonReacher, 2018-08-30:
PX4 1.8.0 - QGC 3.3.2 Android

A timed (30 sec.) loiter was planned just before the landing pattern. The loiter was entered and continued for 4 minutes before I had to manually select the next waypoint to go to. I don't currently have master on my Android tablet to test. I wasn't sure if this was a known issue and/or had been addressed.

I created an issue in QGC and was directed to open an issue in firmware.

https://logs.px4.io/plot_app?log=3ceb70c2-c26b-4c48-818c-0c1c7d31044

dagar, 2018-08-30:
Possibly fixed in https://github.com/PX4/Firmware/pull/8883, but that PR is badly out of date.

dagar, 2018-08-30:
This is another one to add to CI testing once it's been fixed.

closed 2019-02-18 by stale[bot]