PX4/PX4-Autopilot #10377: jMAVSim fails to start

julianoes, 2018-08-31:
With latest master jMAVSim SITL fails to start:

`git bisect` points to 250c5619631f7e836bff9e191d1b03f2478b4ca9

```
SITL ARGS
sitl_bin: /home/julianoes/src/Firmware/build/posix_sitl_default/bin/px4
rcS_path: posix-configs/SITL/init/ekf2
debugger: none
program: jmavsim
model: none
src_path: /home/julianoes/src/Firmware
build_path: /home/julianoes/src/Firmware/build/posix_sitl_default
empty model, setting iris as default
SITL COMMAND: /home/julianoes/src/Firmware/build/posix_sitl_default/bin/px4 /home/julianoes/src/Firmware/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t /home/julianoes/src/Firmware/test_data
INFO  [Unknown] Creating symlink /home/julianoes/src/Firmware/ROMFS/px4fmu_common -> /home/julianoes/src/Firmware/build/posix_sitl_default/tmp/rootfs/etc
65 WARNING: setRealtimeSched failed (not run as root?)

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [Unknown] Calling startup script: bash etc/init.d-posix/rcS 0
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11405132 bytes
INFO  [simulator] Waiting for initial data on UDP port 14560. Please start the flight simulator to proceed..
Buildfile: /home/julianoes/src/Firmware/Tools/jMAVSim/build.xml

make_dirs:

compile:

create_run_jar:

copy_res:

BUILD SUCCESSFUL
Total time: 0 seconds
Options parsed, starting Sim.
Starting GUI...
3D [dev] 1.6.0-pre12-daily-experimental daily

Init MAVLink
INFO  [simulator] Got initial simulation data, running sim..
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
[100%] Built target jmavsim
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet                                                               
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR  [init] Failed loading mixer: etc/mixers/quad_w.main.mix
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
ERROR [Unknown] PX4 daemon not running yet
```

llllIIIIllll, 2018-08-31:
it seems tone_alarm cause the problem
Comment ~/src/Firmware/ROMFS/px4fmu_common/init.d-posix/rcS
line 165
```
tone_alarm start 
```
-->
```
#tone_alarm start
```

bkueng, 2018-08-31:
You're right, I'm working on it. The problem is that we call `exit` in the driver.

bkueng, 2018-08-31:
PR is up: https://github.com/PX4/Firmware/pull/10379

closed 2018-08-31 by bkueng

julianoes, 2018-08-31:
Thanks @bkueng!