PX4/PX4-Autopilot #10381: OFFBOARD mode switches to Manual mode after arming in latest dev firmware

ealdaz, 2018-08-31:
Hello 
We have been doing some tests with pretty recent firmware (master  from 23 August  (790356ef6ddd54b002e4d887a6872fbd270020e1) ) and we have found that an application that was working successfully on 1.8 doesn't work anymore.

We are simply sending mavros/setpoint_position/local  messages to the PX4 thru mavros.

With 1.8 stable we can then switch to OFFBOARD and ARM and system stays in OFFBOARD as expected

With this latest firmware we can switch to OFFBOARD, but when we ARM system exits OFFBOARD and reverts to latest mode (STABILIZE for example)

dagar, 2018-08-31:
Could you share a log?

ealdaz, 2018-09-02:
Of course, I’ll send one thru on Monday.
Thx

Eduardo

ealdaz, 2018-09-03:
Here is a log and a summary of the test actions:

At 6:58 started with PX4 in Stabilized Mode and Disarmed
At approx 7:00  switched to Offboard Mode using RC Transmitter (Channel 7 in our radio)  
At 7:14, armed using QGC. Mode changes from Offboard to Altitude immediately !

Between 7:16 and 7:31 we did some mode changes ( you can ignore)
At 7:32, disarmed.
At 7:40 turned off RC Transmitter

Interestingly with RC transmitter off we could set OFFBOARD and ARM from QGC. After Arming PX4 this time stayed in OFFBOARD mode.

https://logs.px4.io/plot_app?log=d8afda7f-b6ff-4e38-9df9-056489fa838b

This happens also in latest commit d7580aa6

AlexisTM, 2018-09-05:
It seems that all the modes switching were failing with message "Activation failed"

ealdaz, 2018-09-05:
I think this is the same issue I'm experiencing:
https://github.com/PX4/Firmware/issues/10417

chip-pragma, 2019-08-14:
I have same problem on real drone. Firmware version is 1.9.0. What's wrong?

julianoes, 2019-08-15:
That's interesting. I'll try to reproduce this.

julianoes, 2019-08-15:
I can't reproduce this in SITL without RC connection. Looks like I need to try in HITL with RC connected.

julianoes, 2019-08-15:
I can't reproduce this in HITL with RC connected either. Please re-open this issue with clear detailed instructions how to reproduce it, thanks.

closed 2019-08-15 by julianoes

chip-pragma, 2019-08-15:
In SITL works correctly. I will try HITL. 
Also, I updated firmware to 1.9.2. The problem remained on real drone.