PX4/PX4-Autopilot #10389: review PX4 Mavlink module write helper (send_bytes)

dagar, 2018-09-01:
Each mavlink packet is broken up into several writes (header, payload, optional signature). The PX4 mavlink module write helper (send_bytes) checks the remaining buffer on each of these writes and aborts with a logged tx error if there isn't sufficient space.

https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_main.cpp#L942-L994

The result is a garbage partially written mavlink packet and potentially lost data.

I propose we do this check per mavlink message send initially.

LorenzMeier, 2018-09-01:
I need to cross check. It certainly was whole packet size earlier and got broken in the process if it really is broken.

acfloria, 2018-09-01:
I think the messages are splitted. I noticed it when I was writing the IridiumSBD driver. There I am checking if the whole package fits in the buffer and reset that one if it doesn't.

If we are rewriting how packages are sent it would be great if we can get some feedback if the bytes were sent successfully. Currently if one is using `mavlink_msg_MESSAGE_send_struct` there is no feedback if the write is successful. This would be especially helpful for example for low latency telemetry where one wants to make sure that the messages which were scheduled to sent are actually sent.

dagar, 2018-09-01:
https://github.com/mavlink/c_library_v2/blob/master/mavlink_helpers.h#L349

It can be fixed in the mavlink library, but I'd also like to preserve the data and skip the unnecessary work px4 mavlink module side.

julianoes, 2019-02-04:
@dagar should we test this?

dagar, 2019-02-04:
@julianoes yes, there are a number of small mavlink issues like this we should discuss.