PX4/PX4-Autopilot #10391: Switching to HOLD mode using LPE or EKF performs different response

cimbar, 2018-09-01:
When EKF is used as a position estimator with activating both vision position and GPS bitmask, it is possible to switch HOLD mode but in LPE i cannot switch to HOLD mode with GPS and vision position fusion, the fcu rejects auto hold. LPE allows to switch to HOLD mode when only GPS bitmask activated. I cannot combine GPS and vision position in LPE while EKF has this feature. Is this desired feature or just a bug? In both cases, mavros fake_gps module was used and vision position data transferred as fake_gps at the same time.

**Firmware version:** 1.8.0

**Log samples:**
LPE(vision+gps) reject auto hold: https://logs.px4.io/plot_app?log=5fc00993-5f27-4d65-a009-f6f09ca3bf6e
EKF2(vision+gps) switches to hold mode: https://logs.px4.io/plot_app?log=a37b9199-a058-4642-8133-73f1c2a37004

dagar, 2018-09-01:
@TSC21 could you take a look?

aykutkabaoglu, 2018-09-02:
1- landing target was not selected, land detector bitmask was used
2- yes i don't have a GPS hardware and vision position data was sent to fcu both fusing vision data and fake gps together. I was just testing effects of bitmasks so there is no aim to send both data together.

cimbar, 2018-09-02:
1- landing target was not selected, land detector bitmask was used
2- yes i don't have a GPS hardware and vision position data was sent to fcu both fusing vision data and fake gps together. I was just testing effects of bitmasks so there is no aim to send both data together.

cimbar, 2018-09-03:
When i deactivated vision position on LPE and leave fuse gps data bitmask alone, it can switch to HOLD mode, i assumed global position was initialized in that case. I will also try the same issue with LPE_FAKE_ORIGIN.

cimbar, 2018-09-03:
After setting LPE_FAKE_ORIGIN, the problem has been solved. When LPE_FAKE_ORIGIN=1, the device can switch to HOLD mode even only vision positin was fused in LPE.

TSC21, 2018-09-03:
> After setting LPE_FAKE_ORIGIN, the problem has been solved. When LPE_FAKE_ORIGIN=1, the device can switch to HOLD mode even only vision positin was fused in LPE.

It's supposed to work like that, because it creates a virtual global origin from which LPE projects the local position into global coordinates. But it doesn't change the fact that it's not able to change to hold, even when the global position is available from the GPS (with `LPE_FAKE_ORIGIN` unset). It may be a logic error that needs to be solved.

closed 2019-02-13 by stale[bot]