PX4/PX4-Autopilot #10406: [Intel Aero]: Arms but non responsive attitude

santiago3dr, 2018-09-03:
booted, waited for about 2 minutes, successfully armed and motors spun up to idle but copter non responsive to throttle input

https://logs.px4.io/plot_app?log=12385dd2-7b2a-4cff-a7b7-711d50a7a270
https://logs.px4.io/plot_app?log=332bc3b6-2bae-4503-95b2-b4dadda12d42

seeing errors in log [mc_pos_control] Stabilized-Ctrl failed with error: Activation Failed

bkueng, 2018-09-04:
The log doesn't contain any baro data. Do you have another Aero that you can cross-test?

dagar, 2018-09-04:
@santiago3dr when were you last able to fly this vehicle? Do you have the log?

santiago3dr, 2018-09-10:
flashed todays master 33e68b3 and had the same non responsive throttle on the first attempt:
https://logs.px4.io/plot_app?log=4bbc9f6f-74d4-4a7d-bb35-116469d99ba5

ran a few power cycles and got it to work
https://logs.px4.io/plot_app?log=469b7600-3554-414f-bce2-7dbfe9d2e69c

no error messages on qgroundcontrol on unresponsive nor on successful flight

closed 2019-02-13 by stale[bot]