PX4/PX4-Autopilot #10407: Activate drag fusion by default?

RomanBapst, 2018-09-03:
I noticed that I do not get wind estimates on my vtol when hovering. It seems to that the wind estimation based on drag fusion is not active by default.
@priseborough Is there any reason we cannot have this by default? If not then I'm happy to have a look at it.

priseborough, 2018-09-03:
It was not made active by default because:

1) At the time we were not sure how well the default EKF2_BCOEF_*  parameters would work across a range of different airframe types.
2) Other than the baro position error compensation parameters EKF2_PCOEF_* (which are 0 by default and need to be methodically tuned), the PX4/Firmware code for multi-rotor applications did not require wind estimates.
3) Activating the additional states does slightly increase CPU usage

However there is no adverse effect on the EKF estimation because it does not modify any states other than wind so turning it on by default will have no adverse effect other than the small CPU load increase mentioned previously.

dagar, 2018-09-04:
I would double check a few recent logs from a pixhawk or pixhawk 2, but I think we have plenty of margin for a CPU increase.

Typical multicopter log on a pixhawk.
[image]

dagar, 2018-09-04:
Any VTOL specific concerns? For example transitioning to FW which may have an airspeed sensor and airspeed fusion active.

priseborough, 2018-09-04:
The drag fusion method should only be used when hovering. The current ecl would require it to be togged on the PX4/Firmware side of the interface. We could add a check for !_control_status.flags.fixed_wing to line https://github.com/PX4/ecl/blob/master/EKF/control.cpp#L1290

closed 2019-02-13 by stale[bot]