PX4/PX4-Autopilot #10408: Aero optical flow active but not providing local position estimate

rr2-ip, 2018-09-03:
Hi there,

I'm currently trying to get aero optical flow up and running on the Intel Aero RTF drone with a LeddarOne rangefinder.When I start running optical flow, it appears to be working as there are meaningful OPTICAL_FLOW_RAD.integrated_x and OPTICAL_FLOW_RAD.integrated_y values in QGroundControl. The problem is that while optical flow appears to be working, LOCAL_POSITION_NED x and y estimates are constantly zero.

Things I have done to the drone out of the box:

1. Followed this guide:  I have followed [this guide]( https://github.com/intel-aero/meta-intel-aero/wiki/02-Initial-Setup)
2. Flashed PX4 an additional time with v1.8.0 
3. Enabled the rangefinder (SENS_EN_LEDDAR1 = 1)
4. Set the EKF position fusion to use optical flow only in the local position estimate (EKF2_AID_MASK = 2) - Note that GPS is unavailable in the current indoor environment in which I want to fly.
5. Set the primary source of height data of the drone to use the rangefinder (EKF2_HGT_MODE = Range Sensor).
6. Checked that OPTICAL_FLOW_RAD.quality > 100 consistently (approx 150 on average).
7. Checked that the rangefinder is reporting accurate readings.

Does anyone know what I am missing?

FYI, the aero-get-version.py script returns the following:

BIOS_VERSION = Aero-01.00.13
OS_VERSION = Aero Linux Kernel 4.4.113 (can't remember exact name, but it's the one seen [here](https://downloadcenter.intel.com/download/27833/Intel-Aero-Platform-for-UAVs-Installation-Files?v=t)
AIRMAP_VERSION = 1.8
FPGA_VERSION = 0xc2
AeroFC firmware version = 1.8.0

 Regards,

mhkabir, 2018-09-04:
I probably have a fix for this. Stay tuned.

gjwesteneng, 2018-09-11:
Hello,

We are trying to get optical flow working, using the down facing camera and LeddarOne. So far no result, because the LOCAL_POSITION_NED.x and .y stay 0.

Someone from Intel made a test build of the PX4 firmaware where it started working.

The only thing is that fusion of the VISION_POSITION_ESTIMATE has stopped working in this build, which was working in release AeroFC firmware version 1.8.0.

Since our project depends on optical flow in combination with vision pose estimate (coming from our own indoor GPS), we are very eager for any updates on the issue.

Regards,
Gertjan

mhkabir, 2018-09-11:
Possible fix is here : https://github.com/PX4/ecl/pull/505

egauh0113, 2018-09-17:
@mhkabir

adipandas, 2018-09-18:
Hello @mhkabir can you please let me know how to use the fix you suggested [here](https://github.com/PX4/Firmware/issues/10408#issuecomment-420331821)?

mhkabir, 2018-09-18:
Pull Firmware master, switch ECL branch to `pr-of-control-improvements` and build for your target.

JenaEmz, 2019-01-19:
still not working D;

> Pull Firmware master, switch ECL branch to `pr-of-control-improvements` and build for your target.

JenaEmz, 2019-01-19:
> Hello,
>
> We are trying to get optical flow working, using the down facing camera and LeddarOne. So far no result, because the LOCAL_POSITION_NED.x and .y stay 0.
>
> Someone from Intel made a test build [...]

have you solved this problem?

gjwesteneng, 2019-01-21:
We have stopped using LeddarOne since we could not find a branch that gave stable results...

julianoes, 2019-07-09:
@mhkabir do you know what's up here?

LorenzMeier, 2019-07-09:
Sorry, closing again. With this level of disinterest it’s not something we can afford tracking. We are supporting a number of known good setups.

closed 2019-07-09 by LorenzMeier