PX4/PX4-Autopilot #10412: Navigator: Land/RTL quad flys towards home lat/lot/alt

dakejahl, 2018-09-04:
**Describe the bug**
Auto modes `land` and `rtl` cause quad to fly directly towards home lat/lon/alt. This is the third time I've seen this in two weeks. I've seen it ~10 times since the introduction of FlightTasks. We are flying ~7 hours a week, so this is a very rare edge case bug. Logging either dies completely or drops out when this happens. I finally have a partial log (2 minute long dropout) of a flight where this happened. 

**To Reproduce**
1. Takeoff using mode `takeoff`
2. Fly around in position mode
3. Issue `land` or `rtl`
4. This behavior will persist through a land/disarm/arm/takeoff

**Log Files and Screenshots**
https://review.px4.io/plot_app?log=7955d3de-31c9-4434-b30d-cb31fadefb32

**Additional context**
I am pretty sure `takeoff` is causing the bug. I am fairly certain `navigator` gets stuck in a loop somewhere, this would explain the logger dropouts due to thread starvation (navigator priority is higher than logger). I am guessing all `auto` modes will cause this, we only use `land` and `rtl`, but both of those modes do the same thing, fly right towards home.

dagar, 2018-09-04:
I missed the comment about logging dying when you described this earlier. Does that happen 100%?

dagar, 2018-09-04:
@bkueng any thoughts on the logger dropout?

dakejahl, 2018-09-04:
@dagar This is the first time we have gotten a log from a flight with this bug. All of the other flights where this bug popped up the log has either been missing completely, the file is corrupted, or the log just ends immediately after `takeoff`.

dakejahl, 2018-09-04:
[image]

dakejahl, 2018-09-04:
logger_sd_write has pretty much the same max loop time as navigator (70ms difference)

bkueng, 2018-09-05:
Can you try with a different [SD card](https://dev.px4.io/en/log/logging.html#sd-cards)? Which one are you using at the moment?

dakejahl, 2018-09-05:
We're using https://www.oempcworld.com/mm5/merchant.mvc?Screen=PROD&Product_Code=500303&Store_Code=oem&gclid=EAIaIQobChMIsYGl252k3QIVnEwNCh1RlgSFEAYYASABEgLFSvD_BwE

I can get a different SD card and test this... Are you thinking maybe `navigator` is attempting to read a mission that isn't there or something of that nature?

dakejahl, 2018-10-23:
Haven't seen this since September, FlightTasks has seen lots of good work since then. Closing this.

closed 2018-10-23 by dakejahl