PX4/PX4-Autopilot #10415: LIS3MDL driver is missing check_conf() function and check

ksschwabe, 2018-09-05:
It seems that the LIS3MDL driver is missing the "periodic configuration check" that reads out the configuration registers from the device and ensures that the configuration registers are configured correctly and haven't been unintentionally modified by noise on the I2C line. As far as I can tell this feature was "removed" in the refactoring commit 6324a8851dbc94b2bb63f07c7fb29246d3b98aff.

Note: the HMC5883 driver does still have the check_conf() function which is periodically called.

Since the HMC5883/HMC5983 magnetometers are no longer in production, the LIS3MDLs are now being used more commonly, and I suspect we will start to see more "weird" magnetometer issues in the future.

dakejahl, 2018-09-05:
Hi,

> configuration registers are configured correctly and haven't been unintentionally modified by noise on the I2C line

Do you have any data that shows this happening? This seems highly unlikely.

FYI the LIS3MDL has **significant** thermal drift, we had to replace it in our application as the temperature change from moving from inside an air conditioned building to outside in the sun was enough to destroy the calibraton. It manifests itself as moderate to significant toilet bowling.

dagar, 2018-09-06:
@ksschwabe do you have any evidence that this is needed in practice?

zalessio, 2018-09-06:
[image]

These are the plots of the data recorded from the sensor.
First plot are the binary values of magnetic field vector recorded by the LIS3MDL magnetometer.
Second plot is the temperature of the same sensor.
Third plot is the data recorded from the internal magnetometer.

dagar, 2018-09-06:
Would you be able to upload the corresponding flight log? https://logs.px4.io/

zalessio, 2018-09-06:
Sorry I am not. 
We have a custom logging. We have our own format and we are not recording the .ulog .
If I can provide you any other data, just tell me. I am sorry for this!

dakejahl, 2018-09-10:
Could you describe how you are physically connecting to the ASIC? More importantly the power/ground connections. It is likely the sensor is resetting during flight possibly due to a bad solder joint or loose connectors.

dagar, 2018-09-17:
https://github.com/PX4/Firmware/pull/9448

dakejahl, 2018-09-23:
Was this resolved? I am actually curious about what is going on here.

ksschwabe, 2018-10-01:
@dakejahl: Sorry for the delayed response.

Our connectors should be ok; they are JST-GH and I find it unlikely that we have a loose connection.  I implemented a quick and dirty check for the I2C registers, and I did not see this problem for two weeks, but then we had this strange jump in the signals again. Looking at the data, it doesn't seem that the issue was LSB, or MSB readout difference. It could be that the LIS3MDL reset itself, but I haven't been able to reproduce the problem again. It could be that we just have a slightly faulty LIS3MDL, but I haven't had the time to invest in investigate this further for now - I have switched back to a GPS with an HMC5983 on it.

Out of interest, now that the HMC5983 is no longer in production, is there a magnetometer that the Px4 developers recommend/prefer/are using if the LIS3MDL shouldn't be used owing to its temperature related bias drift?

dagar, 2018-10-01:
@ksschwabe could you open a PR restoring the check_conf() functionality?

You could also take a look at the iSentek IST8310 as a replacement for the HMC5983. The PNI RM3100 is also kind of interesting, but probably overkill.

closed 2019-02-04 by stale[bot]