PX4/PX4-Autopilot #10416: could not install the package for px4

snehashahi, 2018-09-05:
When running following command 
sudo -H pip install pyserial empy toml numpy pandas jinja2

I am getting error
Requirement already satisfied: pyserial in /Library/Python/2.7/site-packages
Collecting empy
  Could not fetch URL https://pypi.python.org/simple/empy/: There was a problem confirming the ssl certificate: [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590) - skipping
  Could not find a version that satisfies the requirement empy (from versions: )
No matching distribution found for empy

mhkabir, 2018-09-05:
It seems like it's a pip version issue. 

https://stackoverflow.com/questions/49768770/not-able-to-install-python-packages-ssl-tlsv1-alert-protocol-version

closed 2018-09-05 by mhkabir

snehashahi, 2018-09-06:
I was able to resolve the issue by manually downloading the python packages and installing it.
i dont think its a pip issue as I was able to install one of the packages using pip.