PX4/PX4-Autopilot #10418: Compilation failed : mavlink message ID not declared in this scope

MattMgn, 2018-09-05:
Hi,

I am facing troubles when building px4. I can't explain why I get this error message:

```
matt@matt:~/PX4/Firmware$ make px4fmu-v3_default 
-- PX4 VERSION: v1.8.0-1107-g060463e
-- CONFIG: nuttx_px4fmu-v3_default
-- Build Type: MinSizeRel
-- The ASM compiler identification is GNU
-- Found assembler: /home/matt/PX4/Toolchains/gcc-arm-none-eabi/bin/arm-none-eabi-gcc
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Found PY_jinja2: /usr/lib/python2.7/dist-packages/jinja2  
-- C compiler: arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
-- C++ compiler: arm-none-eabi-g++ (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
-- PX4 ECL: Very lightweight Estimation & Control Library v0.9.0-588-gb861594
-- Building and including px4io-v2
-- Using C++11
-- Release build type: MinSizeRel
-- Adding UAVCAN STM32 platform driver
-- NuttX: px4fmu-v2 nsh cortex-m4
-- ROMFS: px4fmu_common
-- Configuring done
-- Generating done
-- Build files have been written to: /home/matt/PX4/Firmware/build/nuttx_px4fmu-v3_default
ninja: Entering directory `/home/matt/PX4/Firmware/build/nuttx_px4fmu-v3_default'
[9/1049] git submodule nuttx
[10/1049] git submodule ecl
[11/1049] git submodule devices
[12/1049] Building px4io-v2
-- PX4 VERSION: v1.8.0-1107-g060463e
-- CONFIG: nuttx_px4io-v2_default
-- Build Type: MinSizeRel
-- The ASM compiler identification is GNU
-- Found assembler: /home/matt/PX4/Toolchains/gcc-arm-none-eabi/bin/arm-none-eabi-gcc
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Found PY_jinja2: /usr/lib/python2.7/dist-packages/jinja2  
-- C compiler: arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
-- C++ compiler: arm-none-eabi-g++ (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
-- PX4 ECL: Very lightweight Estimation & Control Library v0.9.0-588-gb861594
-- NuttX: px4io-v2 nsh cortex-m3
-- Configuring done
-- Generating done
-- Build files have been written to: /home/matt/PX4/Firmware/build/nuttx_px4io-v2_default
ninja: Entering directory `/home/matt/PX4/Firmware/build/nuttx_px4io-v2_default'
[7/205] git submodule nuttx
[8/205] git submodule apps
[205/205] Creating /home/matt/PX4/Firmware/build/nuttx_px4io-v2_default/px4io-v2_default.px4
[14/1049] git submodule /home/matt/PX4/Firmware/mavlink/include/mavlink/v2.0
[15/1049] git submodule libuavcan
[16/1049] git submodule apps
[599/1049] Building CXX object src/modules/mavlink/CMakeFiles/modules__mavlink_original.dir/mavlink_messages.cpp.obj
FAILED: /usr/bin/ccache /home/matt/PX4/Toolchains/gcc-arm-none-eabi/bin/arm-none-eabi-g++   -DCONFIG_ARCH_BOARD_PX4FMU_V2 -D__DF_NUTTX -D__PX4_NUTTX -D__STDC_FORMAT_MACROS -isystem NuttX/nuttx/include/cxx -isystem NuttX/nuttx/include -I. -Isrc -Isrc/lib -Isrc/modules -I../../src -I../../src/drivers/boards/px4fmu-v2 -I../../src/include -I../../src/lib -I../../src/lib/DriverFramework/framework/include -I../../src/lib/matrix -I../../src/modules -I../../src/platforms -INuttX/nuttx/arch/arm/src/armv7-m -INuttX/nuttx/arch/arm/src/chip -INuttX/nuttx/arch/arm/src/common -INuttX/apps/include -Iexternal/Install/include -I../../mavlink/include/mavlink -fno-common -ffunction-sections -fdata-sections -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -fno-common -ffunction-sections -fdata-sections -g -fno-exceptions -fno-rtti -std=gnu++11 -fno-threadsafe-statics -DCONFIG_WCHAR_BUILTIN -D__CUSTOM_FILE_IO__ -fcheck-new -Wall -Wextra -Werror -Warray-bounds -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wmissing-declarations -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-implicit-fallthrough -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -Wunused-but-set-variable -Wformat=1 -Wno-overloaded-virtual -Wreorder -Wno-format-truncation -fvisibility=hidden -include visibility.h -fno-strict-aliasing -fomit-frame-pointer -fno-math-errno -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -Os -DNDEBUG -MMD -MT src/modules/mavlink/CMakeFiles/modules__mavlink_original.dir/mavlink_messages.cpp.obj -MF src/modules/mavlink/CMakeFiles/modules__mavlink_original.dir/mavlink_messages.cpp.obj.d -o src/modules/mavlink/CMakeFiles/modules__mavlink_original.dir/mavlink_messages.cpp.obj -c ../../src/modules/mavlink/mavlink_messages.cpp
../../src/modules/mavlink/mavlink_messages.cpp: In static member function 'static uint16_t MavlinkStreamTrajectoryRepresentationWaypoints::get_id_static()':
../../src/modules/mavlink/mavlink_messages.cpp:3355:10: error: 'MAVLINK_MSG_ID_TRAJECTORY_REPRESENTATION_WAYPOINTS' was not declared in this scope
   return MAVLINK_MSG_ID_TRAJECTORY_REPRESENTATION_WAYPOINTS;
          ^
compilation terminated due to -Wfatal-errors.
[599/1049] Building CXX object src/modules/mavlink/CMakeFiles/modules__mavlink_original.dir/mavlink_receiver.cpp.obj
FAILED: /usr/bin/ccache /home/matt/PX4/Toolchains/gcc-arm-none-eabi/bin/arm-none-eabi-g++   -DCONFIG_ARCH_BOARD_PX4FMU_V2 -D__DF_NUTTX -D__PX4_NUTTX -D__STDC_FORMAT_MACROS -isystem NuttX/nuttx/include/cxx -isystem NuttX/nuttx/include -I. -Isrc -Isrc/lib -Isrc/modules -I../../src -I../../src/drivers/boards/px4fmu-v2 -I../../src/include -I../../src/lib -I../../src/lib/DriverFramework/framework/include -I../../src/lib/matrix -I../../src/modules -I../../src/platforms -INuttX/nuttx/arch/arm/src/armv7-m -INuttX/nuttx/arch/arm/src/chip -INuttX/nuttx/arch/arm/src/common -INuttX/apps/include -Iexternal/Install/include -I../../mavlink/include/mavlink -fno-common -ffunction-sections -fdata-sections -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -fno-common -ffunction-sections -fdata-sections -g -fno-exceptions -fno-rtti -std=gnu++11 -fno-threadsafe-statics -DCONFIG_WCHAR_BUILTIN -D__CUSTOM_FILE_IO__ -fcheck-new -Wall -Wextra -Werror -Warray-bounds -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wmissing-declarations -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-implicit-fallthrough -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -Wunused-but-set-variable -Wformat=1 -Wno-overloaded-virtual -Wreorder -Wno-format-truncation -fvisibility=hidden -include visibility.h -fno-strict-aliasing -fomit-frame-pointer -fno-math-errno -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -Os -DNDEBUG -MMD -MT src/modules/mavlink/CMakeFiles/modules__mavlink_original.dir/mavlink_receiver.cpp.obj -MF src/modules/mavlink/CMakeFiles/modules__mavlink_original.dir/mavlink_receiver.cpp.obj.d -o src/modules/mavlink/CMakeFiles/modules__mavlink_original.dir/mavlink_receiver.cpp.obj -c ../../src/modules/mavlink/mavlink_receiver.cpp
../../src/modules/mavlink/mavlink_receiver.cpp: In member function 'void MavlinkReceiver::handle_message(mavlink_message_t*)':
../../src/modules/mavlink/mavlink_receiver.cpp:314:7: error: 'MAVLINK_MSG_ID_TRAJECTORY_REPRESENTATION_WAYPOINTS' was not declared in this scope
  case MAVLINK_MSG_ID_TRAJECTORY_REPRESENTATION_WAYPOINTS:
       ^
compilation terminated due to -Wfatal-errors.
[599/1049] Generating nuttx/libc/libc.a
ninja: build stopped: subcommand failed.
Makefile:154: recipe for target 'px4fmu-v3_default' failed
make: *** [px4fmu-v3_default] Error 1
```

Problem seems to come from mavlink, but i don't know why

Any help would be appreciated

matt

dagar, 2018-09-06:
Your mavlink submodule is out of date. Do you have any intentional local changes? If the answer is no, make sure you're synced up with PX4 master, run `make distclean`, and try building again.

MattMgn, 2018-09-11:
Allright, it's now compiling
I was running `make clean `instead of `make distclean`
Thanks
matt

closed 2018-09-11 by MattMgn