PX4/PX4-Autopilot #10419: NXPhlite - FC crashes and does not restart - I2C hardfault

dk7xe, 2018-09-06:
NXPhlite crashes suddenly after a while. Pushing reset button did not resulted in a restart.
Restart happened only after disconnection GPS and power cycling the device.

According David S. that's related due to a I2C hardfault situation.

**Expected behavior**
no crash / restart after reset button pushed

**Log Files and Screenshots**
n/a

**Drone (please complete the following information):**
S500 NXP hovergames Drone

**Additional context**
n/a

davids5, 2018-09-11:
Addressed in https://github.com/PX4-NuttX/nuttx/pull/34, Firmware PR to Follow.

davids5, 2018-09-21:
@dk7xe this fied the QGC connected hardfault:
https://github.com/PX4/Firmware/commit/d2faf5f31d2fbfc4466a5e2c885c2fd445b9861d

This fixed the I2C related hardfault:
https://github.com/PX4/Firmware/pull/10460/commits/b8bc8151ce398ba4a6b04f5c134be31f95c8ce54

Please retest  current master and see if this has been resolved.

dk7xe, 2018-09-24:
Hi @davids5, i did an succesful testflight this morning. Other than with previous version even position hold was working.

closed 2018-09-24 by LorenzMeier