PX4/PX4-Autopilot #10424: Feature: Support RTK and normal GPS blending

AuterionWrikeBot, 2018-09-06:
Change in EKF required:

Currently all gps have to run at the same rate. But RTK (trimble) runs at 20hz and ublox m8n runs at 5hz

priseborough, 2018-09-14:
Prototype here: https://github.com/priseborough/Firmware/tree/pr-gpsBlendRate

PR will be submitted when reversion testing complete. Will require test data for replay testing with RTK 20hz and ublox m8n 5Hz data.

 We should also add user settable parameter to enable running UBlox m8n at 10Hz

AuterionWrikeBot, 2018-09-18:
➤ Wrike Bot commented:

@Paul Riseborough All subtasks have been completed or deferred, you might want to complete this task as well.

closed 2018-10-23 by priseborough

thomasgubler, 2018-10-23:
@hamishwillee could you reach out to @priseborough and make sure we have documentation for this new feature