PX4/PX4-Autopilot #10425: Simulation with pixhawk and flightgear

goodman110110110, 2018-09-06:
The pixhawk firmware has not supported flightgear from version1.5.
But flightgear is an open source software and you can change some hud and gui.
I think if pixhawk new version support flightgear will be a perfect thing.

goodman110110110, 2018-09-06:
I have carried out the auto posctl, and altctl mode in flightgear with firmware 1.4.4.
But it has many bug, for example, it cann't auto takeoff and mission.
If I want to go on with firmware 1.4.4. I will do many works.
Can the new version support flightgear simulation?

julianoes, 2018-09-11:
I don't think this has been used in a while. Contributions would be welcome!

hamishwillee, 2018-09-12:
Yes, it is [not even documented as an option](https://dev.px4.io/en/simulation/hitl.html). The best supported HIL simulation at the moment is probably Gazebo. 

As Julian says, if you can get this working, we'd love to take a PR(s).

clolsonus, 2019-02-14:
I really don't understand the internal px4 data flow or how qgroundcontrol ties in with HIL, but if there is anything I can assist with from the FlightGear interface side, please let me know.  I've successfully interfaced FlightGear with the AuraUAS flight stack and the UMN UAV Lab "Goldy3" flight stack for HIL testing.  Also the UMN UAV lab is in the process of building up a hangar of small UAV models (fixed wing aircraft) that work with FlightGear and JSBSim (standalone).

kaklik, 2020-03-31:
FYI: https://github.com/PX4/Firmware/pull/14539

Jaeyoung-Lim, 2020-05-21:
Closing since flightgear support was added in https://github.com/PX4/Firmware/pull/14539 as pointed out by @kaklik

closed 2020-05-21 by Jaeyoung-Lim