PX4/PX4-Autopilot #10430: Multi-Vehicle Simulation with Gazebo not working: single_vehicle_spawn.launch returns error

tropappar, 2018-09-07:
There seems to be a problem with the launch files and the models of the UAVs. When I follow the example in the developer guide (https://dev.px4.io/en/simulation/multi-vehicle-simulation.html) the launch (roslaunch px4 multi_uav_mavros_sitl.launch) in step 3 fails with following result:

`ERROR [Unknown] Unknown/unhandled parameter: -s
Restoring terminal
process[uav2/iris_2_spawn-8]: started with pid [17474]
[/uav1/mavros:351]: Plugin ftp initialized
[/uav1/mavros:301]: Plugin global_position loaded
process[uav2/mavros-9]: started with pid [17475]
[uav2/sitl_2-7] process has died [pid 17473, exit code 1, cmd <my home folder>/Workspaces/px4/Firmware/build/posix_sitl_default/px4 <my home folder>/Workspaces/px4/Firmware/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -i 2 -d __name:=sitl_2 __log:=<my home folder>/.ros/log/64a1c728-b272-11e8-82c6-3413e83b56c3/uav2-sitl_2-7.log].
log file: <my home folder>/.ros/log/64a1c728-b272-11e8-82c6-3413e83b56c3/uav2-sitl_2-7*.log
`

This problem originates in the single_vehicle_spawn.launch. If I remove the arg -s [...] part, I get the error for the arg -i [...]. If I also remove that part, the launch proceeds without errors but MAVROS cannot connect to the PX4 SITL flight controller.

This was working properly with v1.8.0 but not with the latest version of the master branch.

k2shah, 2018-09-08:
Hi @tropappar,
 I also ran into this issue. The problem seems to stem from two issues caused by the following code 

https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d-posix/rcS#L84-L88

1. The shifting of port numbers causes an overlap (specially on port 14557) which shows up as a bind warning then running the launch file. 

2. The MAV_SYS_ID gets thrown off by 1, not sure if this was an issue, but I needed to remove the +1 (see code linked) in order to get it to work. 

I'm going to perhaps submit a pull request for my fixes, but for the time being feel free to clone my fixes on https://github.com/k2shah/Firmware. I currently can get the `roslaunch px4 multi_uav_mavros_sitl.launch` to spawn two quads that I can arm.

bkueng, 2018-09-11:
@tropappar Can you do a `make clean` and then rebuild the firmware?

@k2shah You're right, there's a port overlapping, and we can change the local port in https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d-posix/rcS#L88 to a higher range.
The MAV_SYS_ID should not be an issue, otherwise we need to change the ID in the launch file.

k2shah, 2018-09-11:
I changed the ID in both the launch file and the rcS file, I was not able to get mavros to connect otherwise. Let me know if I can help, I'm happy to clean up my fixes and submit a PR and attempt to get it to work without changing MAV_SYS_ID

k2shah, 2018-09-12:
any particular range I should set it for? i was thinking having the 
the base UDP ports be 

14550 - 14554 and then offset each by 4*idx so that it can scale without future overlaps, are there any port ranges that I need to use/not use?

hamishwillee, 2018-09-12:
> are there any port ranges that I need to use/not use?

May not be exhaustive, but see: https://dev.px4.io/en/simulation/#default-px4-mavlink-udp-ports

bkueng, 2018-09-13:
> i was thinking having the base UDP ports be 14550 - 14554 and then offset each by 4*idx so that it can scale without future overlaps, are there any port ranges that I need to use/not use?

Yes that makes sense. Since these are the local ports, it doesn't really matter much.

tropappar, 2018-09-14:
@bkueng A make clean helped to solve the issues with the -i and -s args. Thanks for the hint.
After applying the fixes of @k2shah in https://github.com/k2shah/Firmware/commit/43f396e1b5e6af232a5c083ba170e4399f290458 I am able to arm and take off using MAVROS.

tropappar, 2018-09-17:
After some more testing I realized there are two ways to start the PX4 SITL simulation: Either using URDF models or using SDF models. Unfortunately, those two ways don't seem quite compatible. All the comments above are related to the URDF models. Simulation can be started using the launch file multi_uav_mavros_sitl.launch. For the SDF format, simulation can be started using the mavros_posix_sitl.launch BUT it requires the +1 in the MAV_SYS_ID!

Unfortunately, the documentation is very poor on the PX4 SITL simulation, so there is a lot of trial and error work involved here...

k2shah, 2018-09-17:
Yes that's unfortunate, it's hard to pin down what uses the SDF and what uses the scripted UDRF generation. Seems like this is related to why my PR is failing Jenkins. @bkueng can you take a look and help me find what else the rcS file is used for? My guess is that the default ports are hard coded elsewhere too, not sure if i should go down the rabbit hole of changing all the port numbers. 

I agree with you with respect to the documentation, there was some old version of the multi uav sim that used different UDRFs. It was a struggle just to find this bug in the first place. Lets hope we can get it resolved for future users!

k2shah, 2018-09-21:
@tropappar a fix has been pushed. Can close on this?  Thanks to @bkueng  for getting this resolved so quickly!

tropappar, 2019-01-06:
Sorry for the late reply.
It still does not seem to work. I cloned the latest version of the PX4 firmware and added a simple ROS node to each uav namespace in the multi_uav_mavros_sitl.launch launch file.
The code of the ROS node is this one: https://github.com/mavlink/mavros/issues/903#issuecomment-451739072.
When I launch it, the node cannot connect and stays in the loop waiting for FCU connection.
Any ideas?

EdXian, 2019-01-07:
hi @tropappar,
I have an off-topic question. 
I  use the command  `roslaunch px4 multi-posix-sitl.launch` to spawn two UAVs. Then I want to let two drones connect to an object. Then two drones will carry the object cooperatively.
How should I config the launch file?

tropappar, 2019-01-07:
@EdXian 
1. This is not a forum.
2. The launch file you are referring to seems not to be part of this repo.

In principle, you need to add your ROS node to each namespace group in the launch file. Please have a look at this article in the developer guide: https://dev.px4.io/en/simulation/multi-vehicle-simulation.html
You don't need to run the xacro conversion manually.

k2shah, 2019-01-07:
hi @tropappar 
 That's odd indeed, I've just been using commit that I pushed all that while ago. After some digging it looks like the UDP ports were changed in the rcS file.
**https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d-posix/rcS#L84-L88**

Particularly the port number for the simulator `simulator_tcp_port=$((4560+px4_instance))` it used to be `14560`. I tried simply reverting that back but that seems to not work, it also seems like the generic `roslaunch px4 mavros_posix_siti.launch` also fails so it might be a deeper port problem. There's also been a few other changes so I really don't know, maybe @bkueng has some insight? 

Full disclaimer I tested in on a 18.04/melodic so that could be part of the problem, but I was able to get the multi-quad sim working on that machine before fwiw. 

If you want a quick fix/test  you're welcome to pull this https://github.com/StanfordMSL/Firmware/tree/multi-quad
it has just that port fix and nothing else and I just tested it on a 16.04 machine and it spawns 3 quads with the correct connected mavros state.

bkueng, 2019-01-08:
@julianoes recently changed that from UDP to TCP. Julian can you check if that broke multi-vehicle simulation? Might be better to switch back to UDP, I don't see a benefit from the switch.

julianoes, 2019-01-08:
This issue is from September so why do the lockstep changes matter?
I'm open to changing back to UDP but I would like to understand first.

pravardhanreddy, 2019-02-15:
Can you push the changes to the master, the multi uav simulation is currently not working on ubuntu 18.04 melodic

julianoes, 2019-02-18:
@pravardhanreddy I'm not sure what changes you mean. Everything should be working, I think :thinking: . Otherwise, I need some more output and information about "not working".

julianoes, 2019-07-01:
Alright, I'm assuming this is working, otherwise please re-open.

closed 2019-07-01 by julianoes

hamishwillee, 2019-07-02:
FWIW I would assume so too. @lamping7 and others have played with this recently.