PX4/PX4-Autopilot #10434: lidar lite v3 HP can not run in ll40ls drivers

tonight123, 2018-09-07:
**Describe the bug**
A clear and concise description of the bug.

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on '...'
2. Uploaded mission '....' (attach QGC mission file)
3. Took off '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/en/getting_started/flight_reporting.html)).
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.

**Additional context**
Add any other context about the problem here.

tonight123, 2018-09-08:
failed to initialize LidarLiteI2C on busnum=1

tubeme, 2018-09-09:
This is different device than the original v3 device in the way the reading from the sensor is done. Continues v3 vs. cyclic reading v3 HP. So I think you will have to weight until somebody makes new driver for it for PX4 (nuttx).

tubeme, 2018-09-09:
@dagar Can the original driver for the LidarLite v3 be changed so it can handle the V3 HP?

dagar, 2018-09-10:
Someone will need to dig into the manual to determine the differences.

Does anyone have one to test?

tubeme, 2018-09-11:
Here are the two tech docs. 

[LIDAR_Lite_v3_Operation_Manual_and_Technical_Specifications.pdf](https://github.com/PX4/Firmware/files/2372677/LIDAR_Lite_v3_Operation_Manual_and_Technical_Specifications.pdf)

[LIDAR_Lite_v3HP_Operation_Manual_and_Technical_Specifications.pdf](https://github.com/PX4/Firmware/files/2372678/LIDAR_Lite_v3HP_Operation_Manual_and_Technical_Specifications.pdf)

tonight123, 2018-09-17:
ll40ls start pwm ，and I test pwm  Driver not found。

sparky002, 2018-09-26:
It seems the ardupilot forums have a head start into supporting the LIDAR_LITE_v3HP.
Just wanted to bring this to light to potentially help reduce duplicate work on this modification for px4.

https://discuss.ardupilot.org/t/add-support-for-lidar-lite-v3hp-tested-on-pocketpilot/31291

https://github.com/ArduPilot/ardupilot/pull/9169

DanielePettenuzzo, 2018-10-10:
@tonight123 @tubeme this pr: 
https://github.com/PX4/Firmware/pull/10578 was merged last week into master. Now the v3HP works on i2c.

1Gump, 2019-03-11:
Are there any not-so-obvious parameters that need to be set in PX4/Qgroundcontrol(v3.5) to get this working? I could not get the V3HP to respond wired in to a pixhack v5 i2c(3&4), and wouldn't show on the widgets at all. sett the ll04ls to v3hp... i2c. tried both primary and aid . Had to ask owner to verify lidar operation before spending any more time with no response.

hamishwillee, 2019-03-11:
@1Gump Please don't hijack old closed issues. Ask the question on the discussion boards first, and link issues if you think they might be relevant. Docs for lidarlite are https://docs.px4.io/en/sensor/lidar_lite.html
AFAIK there are no other special parameters than the one mentioned there.

hamishwillee, 2019-03-11:
@1Gump Also, could be related to https://github.com/PX4/Firmware/issues/11492

1Gump, 2019-03-11:
>  This message originated outside of DISH and was sent by: noreply@github.com

My apologies. That was my 1st post ever on any of these forums – im not familiar with the protocol / areas yet.

Every Day is an Adventure when nothing is taken for granted.