PX4/PX4-Autopilot #10435: ADS-B and PX4?

tubeme, 2018-09-08:
Hello guys,

We need some help with an ADS-B and the ways we could make it work with PX4.

We saw there is a device ping2020 https://uavionix.com/products/ping2020/#e1b90894089556bc5 that  and only the receiver pingRX https://uavionix.com/products/pingrx/ that wre integrated with the APM and MP but we could not find any info on the PX4 QGC integration.

Is there a way this could be done? If not are there any other ADS-B hardware that can run with PX4 QGC. Maybe something self contained?

Thanks in advance.

Miker1st, 2019-01-03:
Greetings and happy new year.

I would like to also ask this question. We are trying to implement both a gimbal and a PingRX ADS-B device and have found that both devices require the Telem2 port on the Pixhawk 2.1 Cube.

For our application, we would much rather use PX4 than ArduPilot mainly because of how much better the flight characteristics are with PX4 as well as auto pilot capability.

Thank you as well.

LorenzMeier, 2019-01-04:
They don't require TELEM1 as a hard requirement - it is a question of correct configuration.

hamishwillee, 2019-07-08:
FYI, this is supported and docs should be going into user guide in the next couple of weeks.

closed 2019-07-09 by julianoes