PX4/PX4-Autopilot #10436: please help ( get current pitch )

Zvire30, 2018-09-09:
im want write app under PX4fs what will make me simple 1 axis stabilization ( servo ) of camera  . 
But i cant find where i can take current PITCH of drone . 

In struct definton od vehicle_attitude_s is PITCH, but then compiler write me no member and i get olny pitchspeed . Can some one please help me and tell me how get current PITCH of drone from vehicle_attitude message or from what other message can give me those data ? 

thx for answer

mhkabir, 2018-09-09:
vehicle_attitude.q is a quaternion, you can convert it to euler angles and get pitch from there.

closed 2018-09-09 by mhkabir

Zvire30, 2018-09-09:
some example of code please ? im little new . Example will lot of help ;)