PX4/PX4-Autopilot #10450: about baudrate of serial

roseyangyu, 2018-09-11:
I want to control the telem 1 to stop sending mavlink message,I want telem 1 to output the message that is Custom-Defined.And how do I do to config the baudrate of serial4/5?

julianoes, 2018-09-11:
May I ask what piece of hardware you are referring to? That way we can figure out what your requests map to in code.

roseyangyu, 2018-09-12:
I used pixhawk 2,I want to config telem1 and serial4/5.I hope the telem1 don't output mavlink so much so that I can make it output the message that is I need.And the baudrate of serial4/5 how to config?

roseyangyu, 2018-09-12:
In  QGC ,I can't fiind the parameter to config that

bkueng, 2018-09-12:
FYI this will make the UARTs configurable: https://github.com/PX4/Firmware/pull/10337

roseyangyu, 2018-09-12:
I‘m exciting very much! Thank you very much! @julianoes and @bkueng

roseyangyu, 2018-09-12:
but I have not rc.mavlink. I use v1.8.0.

julianoes, 2018-09-12:
> but I have not rc.mavlink. I use v1.8.0.

Same thing there:
https://github.com/PX4/Firmware/blob/v1.8.0/ROMFS/px4fmu_common/init.d/rcS#L604

roseyangyu, 2018-09-13:
I have  config UART7 and UART8 to 115200,but It seems that the baud rate has not changed.It still is 57600.

closed 2018-09-13 by roseyangyu

roseyangyu, 2018-09-13:
config telem1 sucessfully! thank you again

julianoes, 2018-09-13:
@roseyangyu have you done a `make clean` after changing the defconfig?

roseyangyu, 2018-09-13:
oh i'm sorry.thank you very much