PX4/PX4-Autopilot #10457: Unable to arm

lericson, 2018-09-11:
**Describe the bug**
Between v1.6.5 and v1.8.0 there seems to have been quite a lot of changes to the PWM state machine, or I just never delved into this code before. It took me a couple of days but I have narrowed it down to `src/drivers/px4fmu/fmu.cpp` where there is both a publish and a subscribe to the uORB topic `actuators_armed`. It is perhaps intended to be generic to the situation where there is no safety button.

What happens is that the `orb_check()` call on the `_sub_safety` topic never returns an update, for whatever reason. This bug was particularly difficult to find, because mostly everything seemed normal -- and there was no error message, since the logic that is supposed to notify you that you're trying to arm with the safety on is actually not in this same place.

**To Reproduce**
Steps to reproduce the behavior:
1. Run PX4 v1.8.0 as released with Pixracer board, px4fmu v4.
2. Try to start motors
4. Armed and running light comes on (steady red) and PWM signal is at idle.

**Expected behavior**
Motors to spin.

I will forego the rest of the details as I believe this is more than enough information to fix this problem. I'm however not sure why this hasn't happened to you all.

lericson, 2018-09-11:
I don't think my fix is the way to go, but a way around this without diving into why the uORB stuff isn't doing what it should is to simply let `_safety_off = _safety_btn_off` if `GPIO_BTN_SAFETY` is defined, otherwise use the `orb_check()` stuff. This lets the code be agnostic to what type of safety switch is used and yet removes this complication that resulted in a fairly serious bug.

lericson, 2018-09-12:
For completeness, here's the diff I posted to the above two issues:

```diff
commit de177b41bd6293572020858dec441cb974eab9e4
Author: Ludvig Ericson <ludvig@lericson.se>
Date:   Wed Sep 12 18:14:10 2018 +0200

    Set _safety_off = _safety_btn_off if a hardware button exists

diff --git a/src/drivers/px4fmu/fmu.cpp b/src/drivers/px4fmu/fmu.cpp
index a6927563e..23da99ecd 100644
--- a/src/drivers/px4fmu/fmu.cpp
+++ b/src/drivers/px4fmu/fmu.cpp
@@ -1034,6 +1034,8 @@ PX4FMU::run()
 void
 PX4FMU::cycle()
 {
+	bool updated = false;
+
 	while (true) {
 
 		if (_groups_subscribed != _groups_required) {
@@ -1273,10 +1275,11 @@ PX4FMU::cycle()
 			}
 		}
 
-#endif
+		_safety_off = _safety_btn_off;
+
+#else
 
 		/* check safety button state */
-		bool updated = false;
 		orb_check(_safety_sub, &updated);
 
 		if (updated) {
@@ -1287,6 +1290,8 @@ PX4FMU::cycle()
 			}
 		}
 
+#endif
+
 		/* check arming state */
 		orb_check(_armed_sub, &updated);
 
```

santiago3dr, 2018-09-13:
pixracer runs on a flamewheel 450 quad frame
safety switch disabled
950kV motors; dji 420 lite esc's
frsky x8r receiver
leah6 ublox gps
3S lipo

lericson, 2018-09-13:
> @lericson I'd like to move the safety button handling out of px4fmu entirely. #10119

Sounds like a great idea. Perhaps also add some MAVLink log info for safety off/on.

Another thing that has almost cut my fingers up several times now is the fact that you cannot disarm unless the vehicle believes itself to be landed, or you’re in a manual mode. I can see how it makes sense in other contexts but indoors I don’t know. 

- Ludvig

LorenzMeier, 2018-09-13:
I would recommend to use a kill switch for that use case. Disarming is meant for fully tuned, autonomous operation.

davids5, 2018-09-13:
@lericson - would you please test master with #10477 ?

lericson, 2019-02-13:
This was fixed at some point

closed 2019-02-13 by lericson