PX4/PX4-Autopilot #10469: NXPhlite - PWM output does not change after arming

dk7xe, 2018-09-12:
**Describe the bug**
PWM output does not change after arming. Motors don't spin.

**To Reproduce**
1. Power up Hovergames drone
1. press safety switch
3. arm
4. See error

**Expected behavior**
Motors spin with minimum throttle setting

**Log Files and Screenshots**
[image]

**Drone (please complete the following information):**
- NXP Hovergames drone
- NXPhlite rev B
- master branch as of 5th of September

**Additional context**
na

LorenzMeier, 2018-09-12:
@bkueng Could you please have a quick look?

Nielesonimusso, 2018-09-12:
same issue as #10318

lericson, 2018-09-12:
This is also a dup of #10457

lericson, 2018-09-12:
Which I by the way show the error and a fix in

dk7xe, 2018-09-18:
It is working now with `ce31a4` from git.
[image]

closed 2018-09-18 by LorenzMeier