PX4/PX4-Autopilot #10473: Large oscillations in vision-based position control mode (on Intel Aero RTF)

szebedy, 2018-09-12:
**Bug description**
The drone flies with oscillations of 1+ meters around the desired position.

**To Reproduce**
I am using the PX4 release 1.8.0 on the Intel Aero RTF with the following parameter changes in EKF2:
EKF2_AID_MASK: 24 (Don't use GPS, fuse vision position and yaw)
EKF2_EV_DELAY: 20.0 ms
EKF2_HGT_MODE: Vision

Also, for safety reasons I minimized some MPC velocities and accelerations:
MPC_ACC_* = 2.0
MPC_TKO_SPEED = 1.0
MPC_VEL_MANUAL = 3.0
MPC_XY_CRUISE = 3.0
MPC_XY_VEL_MAX = 1.0
MPC_Z_VEL_MAX_UP = 1.0

I am using SVO for visual odometry and send the position estimates through mavros (VISION_POSITION_ESTIMATE).

On the following video you can see the RVIZ visualization of the local position estimate (named drone) and visual odometry output (named drone_vision). They nicely follow each other so the EKF2 estimator seems to be working fine. When the line with the dots appear, the line visualizes the desired trajectory and the dots are the setpoint positions sent to the drone (again with mavros + SET_POSITION_TARGET_LOCAL_NED).

[rviz.zip](https://github.com/PX4/Firmware/files/2376285/rviz.zip)

Any ideas on how this oscillation could be eliminated?

mhkabir, 2018-09-12:
Is your SVO pose estimate correctly scaled?

mhkabir, 2018-09-13:
Are you using an external scale estimation filter?

szebedy, 2018-09-13:
> Are you using an external scale estimation filter?

Not that I know of. SVO is currently only using the camera stream (no IMU, because it only made things worse, maybe because of corrupt timestamps) Could that be the problem? Can PX4 do this, or is this the job of SVO?

ChristophTobler, 2018-09-13:
@szebedy How are you estimating scale if you don't use IMU measurements? I assume you only use the downward -facing camera and not stereo?
If I'm not mistaken SVO alone doesn't use IMU but needs to be coupled to something like https://github.com/ethz-asl/ethzasl_msf to use IMU and get scale.

szebedy, 2018-09-13:
Hi @ChristophTobler thank you for your reply.

I guess I am not estimating the scale. SVO has a parameter called **map_scale**, which is the average scene depth in meters on initialization. I set this to 1.5 and hold the drone at approximately 1.5 meters above the ground when I start SVO.

ChristophTobler, 2018-09-13:
Could you please provide a log?

szebedy, 2018-09-13:
@ChristophTobler, I made a new test because unfortunately I don't have the log from the previous test. This time I tried to initialize SVO more precisely (measured the 1.5 m above the ground such that it corresponds with the map scale) and the oscillation wasn't so large, but still noticeable (about 0.5 m).

Here is the log: https://review.px4.io/plot_app?log=31639308-68b2-4741-a348-ae7d496a3c46
And here is the visualization of the flight: 
[rviz2.zip](https://github.com/PX4/Firmware/files/2380118/rviz2.zip)

Can you tell from these if the oscillation is due to the imprecise VO scaling?

szebedy, 2018-09-16:
Yes the problem was the VO scaling. If I initialize SVO precisely then the drone flies real nicely.

closed 2018-09-16 by szebedy