PX4/PX4-Autopilot #10485: ModuleBase: Multi, Workqueue, Task

dakejahl, 2018-09-15:
I've been working on a new class **`ModuleBaseMulti`** which is exactly what it sounds like. This is pretty necessary to avoid crappy driver hacks to allow multiple instances of an app.

As I have worked through it I am starting to feel like a generic **`ModuleBase`** doesn't seem to fit for both work_queue applications and tasks. There ends up being some extra flash/ram overhead in every single derived class because the base class has functions/variables to support either implementation (task/workq).

It would be fairly straightforward to break this up into **`ModuleBaseTask`** and **`ModuleBaseWork`**, the only consequence is bloating the source code a little bit, but I think it is worth it for (a) the savings in ram/flash and (b) enforcing concrete development standards for derived classes. 

I am suggesting going from a single **`ModuleBase`** to 4 new explicit classes. Again the tradeoff I am making is source bloat for run time efficiency.
**`ModuleBaseTask`**
**`ModuleBaseWork`**
**`ModuleBaseMultiTask`**
**`ModuleBaseMultiWork`**

Thoughts? @dagar @LorenzMeier

dagar, 2018-09-15:
I don't think the flexibility of running something as either a task or in the work queue necessarily needs to be preserved. Aside from not being an issue on the latest platforms, there are other ways we can save memory, including things that would reduce the cost of a task.

dakejahl, 2018-09-16:
Okay. I won't bother with breaking it up, I am submitting a PR for `ModueleBaseMulti` soon, my mpu9250 work will rely on that.

closed 2018-09-20 by dakejahl