PX4/PX4-Autopilot #10489: Cannot enable CAN-Bus in PX4 1.8

Seeelefant, 2018-09-16:
**Describe the bug**
I tried to switch on the CAN bus, however no UAVCAN configuration shown:

**To Reproduce**
Steps to reproduce the behavior:
[image]

This is not a hardware problem, as in ArduPilot this thing is working.

dagar, 2018-09-16:
Which board are you using?

dagar, 2018-09-16:
Through the console (serial or mavlink) could you run `ver all`? The build for older pixhawks that are limited to using the 1st MB of flash only (px4fmu-v2) don't have UAVCAN included. If you have one of the newer chips you can flash the px4fmu-v3 build which contains UAVCAN and more optional peripheral drivers..

Seeelefant, 2018-09-16:
Unfortunately I have the px4fmu-v2 one. Anything I can do here?

dagar, 2018-09-16:
You can build a custom binary (nuttx_px4fmu-v2_default). You'll need to uncomment uavcan here https://github.com/PX4/Firmware/blob/master/cmake/configs/nuttx_px4fmu-v2_default.cmake#L111, but also comment out other drivers and modules you won't be using to make it fit.

Seeelefant, 2018-09-17:
Thanks a lot! Is there any way to determine the board version, before I buy? On ebay its all "Pixhawk PX4 2.4.8" ....

Seeelefant, 2018-09-17:
Hi Dan,
just investigated a little further. 
First it´s a chinese clone board, so ver all gives
[image]
So it says the hardware is V2, whereas the MCU is rev.3. According to https://www.st.com/content/ccc/resource/technical/document/errata_sheet/38/e6/37/64/08/38/45/67/DM00068628.pdf/files/DM00068628.pdf/jcr:content/translations/en.DM00068628.pdf, section 2.2.10, this is fixed for MCU starting with rev 3.

dagar, 2018-09-17:
If you upgrade the bootloader you should be good to go. With a slightly newer bootloader QGC flashing can detect the available flash and choose either px4fmu-v2 or px4fmu-v3.

Recently support was added for doing this on the board itself via parameter. https://github.com/PX4/Firmware/pull/10426
The other option is to build and flash the Bootloader yourself, but you need a jtag/swd debugger. https://github.com/PX4/Bootloader