PX4/PX4-Autopilot #10495: FW altitude control overshoot

jinchengde, 2018-09-17:
FW altitude change, there is overshoot control
[image]

Looking at the pitch control, that is good for setpoint
[image]

the log attach is 
[](https://logs.px4.io/plot_app?log=ba6fc1bf-ef2a-4bd0-b357-7db23e39c1f4)

and also discuss in
 [](http://discuss.px4.io/t/fw-overshoot-when-changing-altitude/2382)

Antiheavy, 2018-09-17:
@RomanBapst @dagar I thought this was a know problem with FW altitude control?  We were able to reduce the overshoot effect slightly by adjusting the TECS tuning, but the pronounced overshoot is still there every time we have a large altitude change.

Here is one of our random flight logs as a reference:
https://review.px4.io/plot_app?log=7c1cf5af-38f4-4f05-8924-8a26f7ac16ac

jinchengde, 2018-09-18:
@RomanBapst @dagar 
I also test in SITL gazebo plane and it also have overshoot during automatic takeoff, here is the video for you reference
[SITL Gazebo Plane](https://youtu.be/U3nGcOyFCQw)

dagar, 2018-09-18:
@priseborough any suggestions?

priseborough, 2018-10-16:
The input shaping applied to the  height and height rate demand inside the TECS _update_height_setpoint function

https://github.com/PX4/ecl/blob/master/tecs/tecs.cpp#L250-L257

can be improved by making the time constant of LPF able to be explicitly set via a parameter. To eliminate overshoot this cannot be much less than the TECS time constant FW_T_TIME_CONST. 

This will need to be done in conjunction with other changes to avoid steady state errors during constant rate climb and descent manouevres, rather than rely on the user adjusting a FW_T_HRATE_FF parameter. 

I will come up with some changes this week.

tubeme, 2018-10-16:
@priseborough Look at what we came upon while tuning the hrate_ff gain.

In 1.8 we see much better response but still the same behavior on manual control.

https://github.com/PX4/Firmware/issues/8941#issuecomment-371755785

priseborough, 2018-10-16:
I've done some Matlab simulation work on an alternative algorithm for the input shaping. It works for ramps and steps, noting that with step inputs, any use of a predictor or integrator term to eliminate height error must result in overshoot as neither TECS or the input shaping filter have future knowledge of when the ramp will level off and the vehicle dynamics cannot perform the infinite acceleration required to do so without overshoot.

Step Demand Example

[image]

Ramp Demand Example

[image]

Ramp Demand Example - Integrator used to eliminate height offset

[image]

The only way we are going to be able to track ramp inputs without height offsets and overshoot is to incorporate move some responsibility for input shaping in the waypoint logic so that TECS is not asked to perform the physically impossible task of tracking a height profile with non continuous first derivative whilst maintaining zero steady state tracking error.

Matlab source code attached:

[height_demand_shaping.m.txt](https://github.com/PX4/Firmware/files/2481539/height_demand_shaping.m.txt)

LorenzMeier, 2018-10-16:
@bresch Could you please contribute here? It sounds like we should try to do what we’re doing for multicopter as well for TECS here.

priseborough, 2018-10-16:
It is also worth noting that lag free tracking of a time varying hight input is most important during approach and flare. At other times in flight, eg climbing or descending between waypoints, some lag is normally not an issue. Other autopilot projects using the TECS algorithm work around this by applying the feed-forward correction during landing approach only. This would be an effective short term solution.

bresch, 2018-10-16:
@priseborough I wrote a library to generate a time-optimal, jerk limited (bang-bang) trajectory based on desired velocity for multicopter velocity control mode (see the [PR](https://github.com/PX4/Firmware/pull/10696)). Position or velocity can be used as the input of the controller and one or several derivative(s) can be used as feedforwards. We could use the same for FW manual control: the input is the desired velocity, based on the sticks.
[image]

I'm planning to do the same for multicopter waypoint navigation (like it's usually done for pick-and-place industrial robots). This could be used to generate a smooth altitude trajectory + feedforwards for TECS in auto mode; do you think it could be a correct solution?

[image]

dagar, 2018-10-16:
> neither TECS or the input shaping filter have future knowledge of when the ramp will level off

In an old PR that was never merged I moved the navigator altitude ramp handling into the FW position controller. https://github.com/PX4/Firmware/pull/8883

bys1123, 2019-05-13:
Is #10746 fix this? @bresch

bresch, 2019-05-13:
@bys1123 Nope, the PR was for multicopter navigation only but the trajectory generator is in a library and could be reused here. 
However, as Paul demonstrated, a simple 2nd order critically damped input filter should fix the issue. If we don't need to have strict requirements on the maximum amplitude of the derivatives, I would just go with that filter.

Antiheavy, 2019-05-14:
>  as Paul demonstrated, a simple 2nd order critically damped input filter should fix the issue. If we don't need to have strict requirements on the maximum amplitude of the derivatives, I would just go with that filter.

curious, is anyone working on this already for fixed wing?

julianoes, 2019-08-15:
@RomanBapst @sfuhrer is this something that could get testing and brought in at some point?

julianoes, 2019-10-09:
@logindiamant your question seems off topic. Please create a new issue with more detail or ask a question in discuss.px4.io.

logindiamant, 2019-10-09:
People, who discussed this main issue publish Thrust value in logs... So, question is on topic, @julianoes

julianoes, 2019-10-09:
Aha sorry @logindiamant. It comes from:
https://github.com/PX4/flight_review/blob/bb39d190b91bb29b5bd43997e26e31976f6b87be/plot_app/configured_plots.py#L331-L333
and therefore:
https://github.com/PX4/Firmware/blob/f9a5c91f8cc5f74ce8d2a0425f7b76d7af65a0a8/msg/actuator_controls.msg#L15

So it's `actuator_controls_0[3]`.

logindiamant, 2019-10-10:
Ooo, many thanks, @julianoes!!!
I created UAV model on nonlinear differential equations with input {roll, pitch, yawrate and thrust}, and just want to verify it on SITL and real fly UAVs. We plan to fuse model in our visual-inertial odometer

bresch, 2022-08-15:
fixed by https://github.com/PX4/PX4-Autopilot/pull/18498

closed 2022-08-15 by bresch