PX4/PX4-Autopilot #10504: Quadcopter PID Tuning

RR2-IP2, 2018-09-18:
Hi,

I am attempting to follow the [PX4 PID Tuning Guide](https://docs.px4.io/en/advanced_config/pid_tuning_guide_multicopter.html), however, I am not sure how to recreate the log plots shown in the guide.

The only plot even roughly close to this that I know how to get is using the Analyze widget in QGroundControl, but I don't know how to find the yaw/pitch/roll setpoint data.  Without the setpoint data, the plots will be significantly less useful in the tuning process. 

I am new to drone development so I am sorry if this is a very fundamental question.  Any help would be greatly appreciated!

Thanks in advance,
Justin

bkueng, 2018-09-18:
Hi Justin

The plots are from [flight review](http://review.px4.io), an online log analysis tool. See https://docs.px4.io/en/getting_started/flight_reporting.html.

RR2-IP2, 2018-09-18:
Perfect, thank you @bkueng

bkueng, 2018-09-18:
Closing it then.

closed 2018-09-18 by bkueng