PX4/PX4-Autopilot #10508: Smart battery data not logged

NadjaKaempf, 2018-09-18:
When looking at the battery data of different logs I realized that the smart battery specific data is not logged correctly.
Different entries ('capacity', 'cycle_count', 'serial_number', 'average_time_to_empty', 'run_time_to_empty') in 'battery_status' store an array of zeros. 

The following log shows this behavior for example:
https://logs.px4.io/plot_app?log=d9bf3e69-ecc5-478c-8713-23581efa84ad

dakejahl, 2018-09-18:
You need to set **BAT_SOURCE** to `1`

dakejahl, 2018-09-18:
Also, make sure you are starting the smart battery driver in your startup script.
e.g `batt_smbus -X start`
This will start the driver on PX4_I2C_BUS_EXPANSION.

dakejahl, 2018-09-18:
Currently it supports starting on PX4_I2C_BUS_EXPANSION and PX4_I2C_BUS_ONBOARD.
@dagar do I need to add options for other i2c buses depending on the hardware?

dagar, 2018-09-18:
For now I think we'll need to copy what other i2c drivers do (try all the buses). See the differential pressure sensors as an example.

As long as the driver can probe and fail gracefully (quickly) you can have it try to start everywhere by default. See rc.sensors for similar examples.

closed 2019-02-13 by stale[bot]