PX4/PX4-Autopilot #10509: open telem 1 to write data

roseyangyu, 2018-09-18:
I want to open("ttyS1",O_RDWR | O_NOCTTY),but it is failed, I can't understand.but if it is ttyS3 and other that is success.and i send 0x0a to serial4/5,but it output 0x0d 0x0a.I don't want 0x0d,how should I do

roseyangyu, 2018-09-18:
int fd= open("/dev/ttyS5",O_RDWR | O_NOCTTY);
            printf("fd = %d",fd);
            cmd_t cmd = {0x81,0x01,0x0a,0x10,0x00,0x32,0x03,0x01,0xFF};
            write(fd,(void *)&cmd,sizeof(cmd));

            close(fd);
 this is code

roseyangyu, 2018-09-18:
It output 0x81 0x01 0x0d 0x0a...How should I do to cut out 0x0d?

roseyangyu, 2018-09-18:
I have comment out the " mavlink start ${MAVLINK_F} " in rcs

roseyangyu, 2018-09-18:
> I want to open("ttyS1",O_RDWR | O_NOCTTY),but it is failed, I can't understand.but if it is ttyS3 and other that is success.and i send 0x0a to serial4/5,but it output 0x0d 0x0a.I don't want 0x0d,how should I do

ttys1 and ttys2 all failed

jinchengde, 2018-10-08:
did you stop mavlink before you use ttys1?
As we know, in rcS, px4 will open ttys1 for mavlink

roseyangyu, 2018-10-08:
> did you stop mavlink before you use ttys1?
> As we know, in rcS, px4 will open ttys1 for mavlink

I have comment out the "mavlink start ${MAVLINK_F} " in rcs.And ttys1 have stopped mavlink

jinchengde, 2018-10-08:
could you test 
mavlink stop-all
at the end of rcS to confirm stop all the mavlink message

roseyangyu, 2018-10-08:
I have input mavlink status,the telemetry is ttys2 only.

roseyangyu, 2018-10-08:
Can you solve that 0x0a and 0x0d ?  I have add the code  but it is no change
    uart_config.c_oflag &= ~(ONLRET | OPOST);
//    uart_config.c_oflag &= OPOST;
    uart_config.c_iflag &= ~(IGNBRK | BRKINT | PARMRK | ISTRIP | INLCR | IGNCR | ICRNL | IXON);
//    uart_config.c_oflag &= ~ONLRET;

@jinchengde,please

jinchengde, 2018-10-08:
so have you try ttyS6 through serial5 or other serial port ?
both the same output?

xdwgood, 2018-10-09:
@roseyangyu 
Target output ： 0x81,0x01,0x0a,0x10,0x00,0x32,0x03,0x01,0xFF
Actual output ： 0x81,0x01,0x0d，0x0a,0x10,0x00,0x32,0x03,0x01,0xFF 
Right？  By the way, the question you asked is not very clear. What you want to ask is that serial port 1 failed to open? Still more output ‘0x0d’?

roseyangyu, 2018-10-09:
I'm sorry,I will mention my question again.
I want to use telem1 to output other message,but open ttys1 is failed.So I change serial5 to output.but output  more than one 0x0d in front of 0x0a.I am more anxious to solve 0x0d than that open ttys1 is failed.

roseyangyu, 2018-10-09:
all port is more than one  0x0d in front of 0x0a @jinchengde

xdwgood, 2018-10-09:
@roseyangyu  maybe help： https://blog.csdn.net/CNHK1225/article/details/77883701

roseyangyu, 2018-10-09:
I try it, hope I can solve it.Thanks@xdwgood

mhkabir, 2018-10-09:
Please use http://discuss.px4.io/ for discussions not related to core flightstack bugs.

closed 2018-10-09 by mhkabir