PX4/PX4-Autopilot #10510: [vio-pipeline] Add Mavlink Odometry loopback

TSC21, 2018-09-18:
Related: https://github.com/PX4/Firmware/pull/9895

Purpose: Similar to what is done with `VISION_POSITION_ESTIMATE` Mavlink message, implement an `ODOMETRY` Mavlink msg loopback.

Concerns: This loopback may serve two purposes:
1. Check the data that is being sent to the FCU - this data is received with the `frame_id` set as `MAV_FRAME_MOCAP_NED` or `MAV_FRAME_VISION_NED`, which means that the data is coming from an external estimator;
2. Get the current odometry estimation from the FCU, which means setting the `frame_id` to `MAV_FRAME_ESTIM_NED` - this would close the pipeline as it should, but requires having a new `vehicle_odometry` msg being propagated on the system, parallel to `vehicle_local_position`, `vehicle_attitude` and `estimator_status`. The consumer for this message would only be the mavlink module, in specific, `mavlink_messages.cpp`, so I don't think it would add a significant overhead to the system.
Another option for this would be creating a `covariances` uORB topic where the pose and velocity covariance matrices would be propagated, and in the mavlink end, add a way to sync the `vehicle_local_position`, `vehicle_attitude` and this `covariances` msg data and propagate this data as an `ODOMETRY` Mavlink message.

Remarks: I don't see a real advantage on 1., but I guess it can be used in parallel with 2. as long as we define a parameter to set what type of loop we want for the odometry.

TSC21, 2018-09-18:
@LorenzMeier, @dagar, @thomasgubler, @mhkabir, @ChristophTobler feedback welcomed. Thanks

thomasgubler, 2018-09-19:
@bkueng can you please share your thoughts here regarding the proposed new message flow?

dagar, 2018-09-19:
Publishing another vehicle_odometry only for mavlink isn't necessarily a problem.

TSC21, 2018-09-19:
@dagar I was trying to analyse it in terms of system overhead.

dagar, 2018-09-20:
It depends on the full usage. Publishing a message really isn't very expensive (a couple hundred bytes of memory total orb overhead + last publish). The issue is if it's logged, at what rate, other modules that use it at all, other modules that poll it, etc.

For example that's why in the initial PR where core messages like vehicle_attitude and vehicle_local_position grew was problematic. There are many copies of each of those structs throughout the system (probably too many already) and we try to log them both at relatively high rates.

bkueng, 2018-09-20:
> a couple hundred bytes of memory total orb overhead + last publish

That's still relevant as it keeps adding overhead to the base system. Not necessarily relevant for this single case, but over time, which is why we hit the RAM limit from time to time.

In this case I'm ok with `vehicle_odometry`.

TSC21, 2018-09-20:
Thanks for the feedback! I will implement the `vehicle_odometry` approach then. It will also be directly used on the RTPS stream, so I think it's a good approach.

closed 2018-12-21 by jgoppert