PX4/PX4-Autopilot #10512: [vio-pipeline] Add vision velocity estimation fusion to EKF2

TSC21, 2018-09-18:
**Purpose:** Get full use of the odometry pipeline by adding linear and angular velocity fusion to EKF2.

**Concerns:** @priseborough maybe you want to add some details on how can this be achieved?

TSC21, 2018-09-18:
@mhkabir FYI.

TSC21, 2019-01-29:
@jkflying FYI

v01d, 2019-03-20:
Hi, having this issue open means that currently the odometry message does not fuse the velocity? 
Also, is it possible/planned to support velocity-only vision estimate input so that velocity-only offboard control is performed?

v01d, 2019-05-23:
Now that 1.9.0 is close to being released I would like to re-ask my previous question, in case situation is different. I'm trying the RC and it would seem that the velocity component from the odometry from VIO is not used. Is this supported in any other way?

TSC21, 2019-05-23:
> Now that 1.9.0 is close to being released I would like to re-ask my previous question, in case situation is different. I'm trying the RC and it would seem that the velocity component from the odometry from VIO is not used. Is this supported in any other way?

Still open for development. If you feel you can give a contribution, I will be happy to review and bring others to take a look as well.

TSC21, 2019-05-23:
> so that velocity-only offboard control is performed?

Note: this has nothing to do with the fact that EKF2 fuses vision velocity estimates or not. You can use velocity control by closing the loop on the offboard side (knowing the position).

TSC21, 2019-08-21:
Still relevant

mhkabir, 2019-11-20:
Fixed.

closed 2019-11-20 by mhkabir