PX4/PX4-Autopilot #10514: Adding new rotation (SENS_BOARD_ROT) gives irrelevant orientation

a6a3uh, 2018-09-18:
**Describe the bug**
As roll 90 yaw 270 orientation is missing for SENS_BOARD_ROT I added as per  this [commit](https://github.com/a6a3uh/Firmware/commit/6a0ddf1c72be9e46ce0cfebf7ec15e61776c4f89). And found resultant orientation irrelevant and even non orthogonal. 

**To Reproduce**
Steps to reproduce the behavior:

To nail problem down I commented my code as you can see in the commit above and copy-pasted code from roll 270 yaw 270. 

1) I set `SENS_BOARD_ROT` to 26 (`ROTATION_ROLL_270_YAW_270`) and ensure proper HUD readings
2) I change to `SENS_BOARD_ROT` to 37 (which uses the same exactly code) and I see HUD starts drifting away.
3) I reproduced the behaviour in `sitl_gazebo` as well. Look at [firmware](https://github.com/a6a3uh/Firmware/commit/41cc3ac487d55ac037b363f428bf15bfc540cc42) and  [sitl_gazebo](https://github.com/a6a3uh/sitl_gazebo/commit/7be651bd8061c2cecbbd353417a460748884f99d) commits. I modified `iris` and `standard_vtol` models there as well as init scripts to set orientation roll 270 yaw 270. `iris` works the same way as described in p.2. 
4) `standard_vtol` shows some very strange behaviour as it not works even with `SENS_BOARD_ROT`=26. 
while `iris` readings are ok:

```
pxh> listener sensor_accel -i 0

TOPIC: sensor_accel instance 0 #1
 sensor_accel_s
        timestamp: 14384724  (0.002421 seconds ago)
        error_count: 0
        device_id: 6789478 (Type: 0x67, UNKNOWN:12 (0x99))
        x: -0.2478
        y: 10.0237
        z: -0.1775
        integral_dt: 2539
        x_integral: -0.0008
        y_integral: 0.0255
        z_integral: -0.0004
        temperature: 32.0000
        scaling: 0.0000
        x_raw: 0
        y_raw: 0
        z_raw: 0
```

`standard_vtol` readings are irrelevant:

```
pxh> listener sensor_accel -i 1

TOPIC: sensor_accel instance 1 #1
 sensor_accel_s
        timestamp: 91892223  (0.003203 seconds ago)
        error_count: 0
        device_id: 0 (Type: 0x00, UNKNOWN:0 (0x00))
        x: 12.2135
        y: 0.1455
        z: -0.2555
        integral_dt: 0
        x_integral: 0.0000
        y_integral: 0.0000
        z_integral: 0.0000
        temperature: 0.0000
        scaling: 0.0000
        x_raw: 0
        y_raw: 0
        z_raw: 0
```

**Expected behavior**
As code for `SENS_BOARD_ROT` = 26 and `SENS_BOARD_ROT` = 37 is the same I'm expecting nothing should change if I switch between those. Or selecting `SENS_BOARD_ROT` = 37 should give same HUD readings as if `SENS_BOARD_ROT` = 26.

**Additional context**
I started discussion in [forum](http://discuss.px4.io/t/pixhawk-orientation-roll-90-yaw-270-is-absent-in-the-list/7796) with no great success as well as in slack with same results.

a6a3uh, 2018-09-18:
tested on **pixhawk 4** and **sitl_gazebo**.

a6a3uh, 2018-09-19:
My bad. I've missed that. I'll test it hope this will solve my issue.
Still `standard_vtol` behaviour in SITL is strange. As it outputs acceleration X = g with Roll 270 Yaw 270. While correct behaviour is Y = g like `iris` shows. But this is probably an issue for `sitl_gazebo` project.

TSC21, 2018-09-19:
That's not an issue on `sitl_gazebo`. It has to be with the coordinate frame of reference and it is well documented.

a6a3uh, 2018-09-19:
Dear Nuno Marques. Kindly could you pls point me to what I'm missing while setting rotation for `standard_vtol`?

I tried to apply the same rotation roll 270 yaw 270 for both `iris` and `standard_vtol` models as you can see in [this](https://github.com/a6a3uh/sitl_gazebo/commit/7be651bd8061c2cecbbd353417a460748884f99d) commit. And get correct behavior with `iris` model while `standard_vtol` shows irrelevant orientation.

I take into account different frames used in px4 (forward-right-down) and gazebo (forward-left-up).

a6a3uh, 2018-09-19:
Also [enum](https://github.com/PX4/Firmware/blob/a5d5d5e38ea946abce254c2f2886307a76f907a9/src/lib/conversion/rotation.h#L51) with possible rotations `SENS_BOARD_ROT` in `px4` uses the same orientations from 0 up to 25 as `mavlink`'s [MAV_SENSOR_ORIENTATION](https://github.com/mavlink/c_library_v2/blob/d8fcf0a694dc11b3f83b89a0970e3d8c4e48d418/common/common.h#L789). Starting from 26 rotations are different. Maybe it worth a feature request to make rotations in `px4` and `mavlink` consistent.

a6a3uh, 2018-09-19:
> Did you update the table as well? https://github.com/PX4/Firmware/blob/master/src/lib/conversion/rotation.h#L96-L132

[Implemented](https://github.com/a6a3uh/Firmware/commit/7ca2191401898502a6487d9cdc4cd9c9ebb7dfe5) and tested with real hardware. All is working fine now. Thanks a lot! [Tested](https://github.com/a6a3uh/sitl_gazebo/commit/d993efe850b957f83a87bee59bbd131b920a7081) as well with `sitl_gazebo`. `iris`model is working well with rotated IMU.

closed 2019-01-31 by mhkabir