PX4/PX4-Autopilot #10517: Px4 SITL + QGC - critical navigation failure - check sensor calibration

geevargs, 2018-09-18:
Every time I run PX4 SITL and send waypoint commands in QGC or using python api which get visualised using Airsim.
The connection to Airsim is established and it deos basic commands like Takeoff ,Landing,Hold,Loiter...
But when I try to execture waypoints or orbits,surveys gets 
ERROR CRITICAL NAVIGATION FAILURE - CHECK SENSOR CALIBR  
WARN  Failsafe enabled: no local position 
WARN  Failsafe mode enabled.
WARN  [ecl/EKF] EKF GPS fusion timeout - reset to GPS
WARN  [commander_tests] Failsafe enabled: no global position

Steps to reproduce the behavior:
1. cd PX4/Firmware and make sudo make posix_sitl_default  none
2. QGC simple Goto waypoint or orbit at position or Python client in Airsim
3. Takes off as it always deos but fails after that.
4. which causes the failure.

**Expected behavior**
It would follow control input and follow waypoints and formation flights.

[image]

[image]

https://logs.px4.io/plot_app?log=ea723e21-e9a5-40bc-a1a8-c23995b00210

TEST conditions:
Linux 16.04 and Airsim 4.18
QGroundcontrol v3.4.3
Python 2.7.12

SusejLav, 2019-09-16:
I'm currently having the same issue, i'm using the default settings in AirSim and using these versions:

Windows 10 and Airsim 1.2.0, PX4 1.8.2
QGroundcontrol v3.4.3
Python 3.6

julianoes, 2019-09-17:
@geevargs and @SusejLav does this have to do with CPU load which would lead to timeout? If so it will be fixed when 1.9 with lockstep is supported by Airsim.

SusejLav, 2019-09-17:
I don't know if it's due to CPU, i'm trying to get gps status in PX4 console but i get a "INFO [gps] not running" message.
Also, i never get the home position log message, i start the PX4 and wait for around 15 seconds but the most i get is this:
[image]

Additionally, i can't get the 'pos' command to work for me, if i run it i get a "Invalid command: pos" message.

julianoes, 2019-09-24:
The `gps` module/driver is not running in SITL. Instead, GPS data is arriving via MAVLink and received and published in the `simulator` module.

julianoes, 2020-01-30:
Lockstep is now working with Airsim, closing this. Please re-open if the problem persists.

closed 2020-01-30 by julianoes

bigbellmercy, 2020-10-29:
> Every time I run PX4 SITL and send waypoint commands in QGC or using python api which get visualised using Airsim.
> The connection to Airsim is established and it deos basic commands like Takeoff ,Landing,Hold,Loiter...
> But when I try to execture waypoints or orbits,surveys gets 
> ERROR CRITICAL NAVIGATION FAILURE - CHECK SENSOR CALIBR
> 
> WARN Failsafe enabled: no local position 
> WARN Failsafe mode enabled.
> WARN [ecl/EKF] EKF GPS fusion timeout - reset to GPS
> WARN [commander_tests] Failsafe enabled: no global position
> 
> Steps to reproduce the behavior:
> 
> 1. cd PX4/Firmware and make sudo make posix_sitl_default none
> 1. QGC simple Goto waypoint or orbit at position or Python client in Airsim
> 1. Takes off as it always deos but fails after that.
> 1. which causes the failure.
> 
> **Expected behavior**
> It would follow control input and follow waypoints and formation flights.
> 
> [image]
> 
> [image]
> 
> https://logs.px4.io/plot_app?log=ea723e21-e9a5-40bc-a1a8-c23995b00210
> 
> TEST conditions:
> Linux 16.04 and Airsim 4.18
> QGroundcontrol v3.4.3
> Python 2.7.12
> 
> 

Hi! Was it solved? I have the same error message when I use external vision using Vicon.

m1baldwin, 2020-11-18:
@bigbellmercy I have no idea if it was ultimately solved, however I am seeing something similar now in my setup. I am running the following:
Linux 18.04
Airsim: Latest Master as of Oct 7th (commit: d59ceb7f63878f5e087ea802d603ba0fd282ff56)

I am trying to simulate a drone which is being fed vision pose estimate from a visual slam algorithm running on the camera feed from the simulated cameras on board the drone. I have configured the px4_sitl process similar to how it would be configured in a lab environment using an external vicon system. I am using the elf2 estimator with flags set to fuse in vision pose and vision yaw.

When I try taking off, the drone takes off, but moments later will start to drift and fly off. I have noticed that the position estimate from the visual slam algorithm is of good quality and not failing, even when the drone is starting to fly off. I get the messages:
INFO  [ecl/EKF] 199431178: EKF commencing external vision yaw fusion
INFO  [ecl/EKF] 199497201: EKF commencing external vision position fusion
INFO  [commander] Takeoff detected
ERROR [commander] Critical navigation failure! Check sensor calibration
WARN  [mc_pos_control] Position-Control Setpoint-Update failed
WARN  [mc_pos_control] Failsafe: blind land
WARN  [ecl/EKF] 268276504: EKF ev hgt timeout - reset to baro
INFO  [ecl/EKF] 270220709: EKF External Vision Data Stopped
WARN  [ecl/EKF] 270226709: EKF stopping navigation

I am running the simulation in lockstep using the tcp interface between arisim and the px4_sitl process, so I don't think the above discussion about this fixing the issue is actually working.

bigbellmercy, 2020-11-18:
@m1baldwin That's too bad. The 'Critical navigation failure' message and your symptom similar to my case happen repeatedly to my **real** indoor flight with EKF2 + external vision with Vicon or VIO. Yesterday by this problem, a drone soared up and crashed on the ground (sadly).
Though I thought that geomagnetic distortion in that area cause it, it is not because it happens in simulation environment as your case. 
We'd rather report it first to @julianoes

bigbellmercy, 2020-11-19:
It seems a bug because the error "Critical navigation failure! Check sensor calibration" happens when the gps velocity and position innovations (errors) are too large when I checked the codes. But when external vision by Vicon or VIO is used, gps is not used. So, this error check should be disabled automatically when gps is not used but it's not currently.

bigbellmercy, 2020-11-19:
@m1baldwin How about detach GPS module in your simulation? I'll do it too in real flight.

bigbellmercy, 2020-11-24:
@m1baldwin [Good news] Since I detached the GPS module, the error does happens. So, my thought above seems correct.

blandry, 2020-11-24:
@julianoes just FYI I think this is an actual issue. We kept getting the "Critical navigation failure" running v1.11 and it disappeared after switching back to v.1.9.2. Pos control ran flawlessly so those checks are either broken or too sensitive? (didn't look at what they did specifically)

HarshilNaikk, 2021-02-02:
> @bigbellmercy I have no idea if it was ultimately solved, however I am seeing something similar now in my setup. I am running the following:
> Linux 18.04
> Airsim: Latest Master as of Oct 7th (commit: d59ceb7f63878f5e087ea802d603ba0fd282ff56)
> PX4: v1.10.0
> 
> I am trying to simulate a drone which is being fed vision pose estimate from a visual slam algorithm running on the camera feed from the simulated cameras on board the drone. I have configured the px4_sitl process similar to how it would be configured in a lab environment using an external vicon system. I am using the elf2 estimator with flags set to fuse in vision pose and vision yaw.
> 
> When I try taking off, the drone takes off, but moments later will start to drift and fly off. I have noticed that the position estimate from the visual slam algorithm is of good quality and not failing, even when the drone is starting to fly off. I get the messages:
> INFO [ecl/EKF] 199431178: EKF commencing external vision yaw fusion
> INFO [ecl/EKF] 199497201: EKF commencing external vision position fusion
> INFO [commander] Takeoff detected
> ERROR [commander] Critical navigation failure! Check sensor calibration
> WARN [mc_pos_control] Position-Control Setpoint-Update failed
> WARN [mc_pos_control] Failsafe: blind land
> WARN [ecl/EKF] 268276504: EKF ev hgt timeout - reset to baro
> INFO [ecl/EKF] 270220709: EKF External Vision Data Stopped
> WARN [ecl/EKF] 270226709: EKF stopping navigation
> 
> I am running the simulation in lockstep using the tcp interface between arisim and the px4_sitl process, so I don't think the above discussion about this fixing the issue is actually working.

@m1baldwin i am facing the exact same issue, but in gazebo SITL. I added a custom model, and was tuning the gains for the Roll and Pitch, but this always keeps messing things up. If this has been solved, any suggestions?

unravelwool, 2021-05-25:
> @m1baldwin [Good news] Since I detached the GPS module, the error does **not** happen. So, my thought above seems correct.

Hi, I have met the same problem. Could you tell me how to detach the GPS module. Thank you in advance :)

bigbellmercy, 2021-05-26:
@unravelwool For my case the error happened in real drone not in simulation. I disconnected the GPS cable from Pixhawk. And changed GPS-use related parametes such as GPS_1_CONFIG: Disabled, MAV_1_CONFIG: Disabled

MuhammadBilal1, 2021-06-04:
Hi Everyone,
I am also getting the same issue with my real drone. Anyone found the solution to this problem?

I started the **MISSION** flight and it tried to flew for a few (8-10) seconds, and then the following happened:
0:8 Critical Navigation Failure
0:10 Check Sensor Calibration
0:11 Fail Safe Enabled

0:12 **Touched Ground**

0:13 No Local Condition
0:14 ___ Flight Mode
0:20 Disarmed
0:21 Mission Flight Mode
0:56 Communication Lost
0:58 Communication Regained 

**Here are the details of my vehicle:**
1- QuadCopter (4 Arms, 8 Motors)
2- PixHawk 4 by Holly Bro
3- PX4 (Latest Stable: 1.11.3)
4- QGroundControl (Latest version as of writing this comment)
5- Windows 10 & Ubuntu 20.04
6- GPS Module is also attached
7- All Calibrations were done with QGroundControl
8- RFD900+ (Telemetry)

julianoes, 2021-08-30:
@MuhammadBilal1 I suggest to create a new fresh issue with what you found, and please attach a link to the logfile that you upload to logs.px4.io. Otherwise it's impossible to see what is going on.

MuhammadBilal1, 2021-08-30:
> @MuhammadBilal1 I suggest to create a new fresh issue with what you found, and please attach a link to the logfile that you upload to logs.px4.io. Otherwise it's impossible to see what is going on.

Thank you for coming back on this issue.
Actually, my issues were fixed.

**What was the problem?**
There was a little magnet attached to the body of the drone on purpose, and that was near to the GPS.
Removing that magnet fixed my issue.

BV-OpenSource, 2021-11-25:
@HarshilNaikk @m1baldwin We kinda fixed that problem by changing the `real_time_update_rate` on Gazebo world.
Our file looks like this:
```<?xml version="1.0" ?>
<sdf version="1.5">
  <world name="default">
    <!-- A global light source -->
    <include>
      <uri>model://sun</uri>
    </include>
    <!-- A ground plane -->
    <include>
      <uri>model://ground_plane</uri>
    </include>
    <physics name='default_physics' default='0' type='ode'>
      <gravity>0 0 -9.8066</gravity>
      <max_step_size>0.004</max_step_size>
      <real_time_factor>1</real_time_factor>
      <real_time_update_rate>250</real_time_update_rate>
      <magnetic_field>6.0e-6 2.3e-5 -4.2e-5</magnetic_field>
    </physics>
  </world>
</sdf>
```

After this change, didn't have any problem.