PX4/PX4-Autopilot #10518: logger Check assertion failure for other ramification.

davids5, 2018-09-18:
**Describe the bug**

I found this on NXPHlite-v3 while debugging hardfaults. 

**Assertion failed at file:semaphore/sem_holder.c line: 915 task: logger**

This needs to be checked, it may be nothing because there is a reboot.

[image]

**To Reproduce**
1. Enable CONFIG_DEBUG_FEATURES and CONFIG_DEBUG_ASSERTIONS
```
-# CONFIG_DEBUG_FEATURES is not set
+CONFIG_DEBUG_FEATURES=y
+
+#
+# Debug SYSLOG Output Controls
+#
+CONFIG_DEBUG_ERROR=y
+CONFIG_DEBUG_INFO=y
+# CONFIG_DEBUG_WARN is not set
+CONFIG_DEBUG_ASSERTIONS=y
```
2. type reboot on console
3. See error
*I had to remove the DBUGASSERTS in the gpio driver to get the system running.

**Expected behavior**
This may be fine as the reset will clean up the system.

davids5, 2018-09-18:
These needed removal to test as they art overzealous
```
diff --git a/arch/arm/src/kinetis/kinetis_pingpio.c b/arch/arm/src/kinetis/kinetis_pingpio.c
index b887920816..76abf0db53 100644
--- a/arch/arm/src/kinetis/kinetis_pingpio.c
+++ b/arch/arm/src/kinetis/kinetis_pingpio.c
@@ -69,9 +69,6 @@ void kinetis_gpiowrite(uint32_t pinset, bool value)
   unsigned int port;
   unsigned int pin;
 
-  DEBUGASSERT((pinset & _PIN_MODE_MASK) == _PIN_MODE_GPIO);
-  DEBUGASSERT((pinset & _PIN_IO_MASK) == _PIN_OUTPUT);
-
   /* Get the port number and pin number */
 
   port = (pinset & _PIN_PORT_MASK) >> _PIN_PORT_SHIFT;
@@ -113,8 +110,6 @@ bool kinetis_gpioread(uint32_t pinset)
   unsigned int pin;
   bool         ret = false;
 
-  DEBUGASSERT((pinset & _PIN_MODE_MASK) == _PIN_MODE_GPIO);
-  DEBUGASSERT((pinset & _PIN_IO_MASK) == _PIN_INPUT);
 
   /* Get the port number and pin number */
 

```

bkueng, 2018-09-19:
A quick code inspection did not reveal anything: waiting threads are woken up and stopped before a semaphore/mutex is destroyed.
Could it be that you ran in a timeout due to the debugging?

davids5, 2018-09-19:
@bkueng - Thanks for looking.

> Could it be that you ran in a timeout due to the debugging?

I do not think so. I can have a closer look after I figure out the hard fault from hell!

julianoes, 2019-07-09:
@davids5 what's the status here?

davids5, 2019-07-09:
@julianoes - there was a recent change in nuttx that should be back ported. It fixes this.  It is it a super low priority and we will get it on the next uptake. So let's close this unless it is getting in someone else's way.

closed 2019-07-09 by julianoes