PX4/PX4-Autopilot #10537: Udefined reference preventing cross compiling

pbafff, 2018-09-21:
I'm trying to cross compile on Ubuntu 18.04 for the BeagleBone Blue. The process almost completes but fails due to the error : 
`src/lib/DriverFramework/framework/src/libdf_driver_framework.a(I2CDevObj.cpp.o): In function `DriverFramework::I2CDevObj::_readReg(unsigned char, unsigned char*, unsigned int)':
/home/andre/Firmware/build/posix_bbblue_cross/../../src/lib/DriverFramework/framework/src/I2CDevObj.cpp:174: undefined reference to `rc_i2c_read_data'`
I installed librobotcontrol to my development computer so I don't understand why it's saying this function is undfined.

pbafff, 2018-09-23:
Tried to compile on the Beagle itself and ran into the same error.

UAV-Pilot, 2018-09-24:
@pbafff  I thought I documented necessary steps on: https://docs.px4.io/en/flight_controller/beaglebone_blue.html. If one follows those steps, and still encounters issues, then I might have missed something. Anyway, did you follow the steps in the above page?

closed 2018-10-01 by pbafff

PRASANTHCUSAT, 2019-05-10:
Hi, I'm new to PX4. As per the above documentation, I have tried both native and cross builds for beaglebone blue. But I cannot find the make rules for the target **posix_bbblue_cross**, which the documentation says for building. Here are some additional informations:
Running Debian Image 2018-10-07 in Beaglebone Blue
gcc version 6.3.0 20170516 (Debian 6.3.0-18+deb9u1) in Beaglebone Blue
gcc version 6.1.1 20160711 (Linaro GCC 6.1-2016.08) in host system (ubuntu 18.04) for cross build (version lower than gcc in target board as per the documentation).

Please help me as I was trying for a week now.