PX4/PX4-Autopilot #10538: UART TX DMA

ksschwabe, 2018-09-21:
Currently it looks like there is only UART DMA on the RX line.

We were trying to implement a "fast" file transfer via UART to a UART to Ethernet bridge and discovered that we were running into CPU usage issues. It looks to us that only the UART RX makes use of DMA. It should be a lot more computationally efficient to make use of the UART TX DMA as well for sending of large file.

Perhaps @dagar and @davids5 know more about the background working of the NuttX UART sending.

mhkabir, 2018-09-21:
@bkueng @davids5

davids5, 2018-09-21:
The issue is the DMA on the stm32 is to make up for a lack of RX FIFO in HW. 
First Question is:If there are HW DMA resources available.
Then adding the same for the TX side will help so a point. But the real win is streaming DMA. This is only partially (incomplete) in Nuttx and would need some significant development effort.

dagar, 2019-06-24:
Upstream NuttX pull request (that's stalled). https://bitbucket.org/nuttx/nuttx/pull-requests/858

davids5, 2019-06-25:
@dagar I told Greg it is on my todo list during the H7 port. However we will just do the circular mode.

mhkabir, 2019-09-23:
Go away

julianoes, 2019-09-25:
Seriously, we can't all waste our time yelling at a bot.

hamishwillee, 2019-09-25:
@julianoes SURE WE CAN :-)

davids5, 2019-12-01:
F7 is in test now.

davids5, 2020-02-28:
Up-streamed

closed 2020-02-28 by davids5