PX4/PX4-Autopilot #10539: Make the MC Position Controller transition aware

RomanBapst, 2018-09-21:
**Description**
Multicopter position controller ignores VTOL transitions. This is a problem because position setpoints after a transition do not get reset and as a result the vehicle will navigate back into the direction of the last waypoint before continuing.
Furthermore, when the vehicle is in fixed wing mode there should not be any flighttask running.

@MaEtUgR @Stifael What's the best way to make the position controller transition aware? Basically how it used to work was that the commander set the flag_control_position to false during a transition.
https://github.com/PX4/Firmware/blob/master/src/modules/commander/Commander.cpp#L3264

So during transitions the position controller would only control altitude which is what we want.

I guess the easiest way would be to create a transition flighttask which generates the altitude setpoint.
Any input from your side?

tubeme, 2018-09-22:
We've had this "navigate back into the direction of the last waypoint before continuing." on couple of occasions. So good idea to make some changes here.

dagar, 2018-09-23:
I see a spectrum of options here. 

1. One possible way out of this I'd like to explore is moving away from the VTOL state being this orthogonal state expected to work during any mode at any time. Instead we could think about handling it as something more like a position setpoint type. This would provide all the necessary structure for moving in and out of transitions cleanly. The actual commands could come from a VTOL module.
2. Handle FW & MC separately in what would more or less be an OFFBOARD control mode. They'd both listen to the same offboard setpoint, but MC would be altitude only, etc.

tubeme, 2018-09-24:
@dagar @Stifael  Important notes on this change because it might lead to fundamental changes in the transition. And the transition is very stable now so we should touch with extreme caution.

About p.2 Up to the end of transition in case no airspeed sensor or up to reaching transition airspeed VT_ARSP_TRANS, the MC should handle most of the attitude, altitude and position control, because there is no lift in the foil so the control surfaces of the airplane does not produce much.

Despite there is no lift early in the transition the MC and FW controllers should work in sync and both produce control despite the airplane control surfaces will not be effective. In case of gusty and heavy weather this synchronous work of both produces better results.

This is handled by blending speed when there is airspeed sensor. If there is no airspeed sensor, the blending is all the time I think.

BTW I remember the back transition at some point had a severe problems due to not resetting the MC POS controller prior transition. Cannot remember which Issues or Pull requests handled this.

thomasgubler, 2018-09-26:
[image]

sanderux, 2018-09-26:
Should be located here: https://github.com/PX4/Firmware/blob/master/src/modules/navigator/mission.cpp#L818-L848

Last known working as far i know https://github.com/PX4/Firmware/tree/DQRC12F

RomanBapst, 2018-10-16:
> How was this done in the past? I am asking because the output of the position controller is an attitude + throttle.

@Stifael In the past the position controller worked based on control flags (vehicle_control_mode) and not directly based on the navigation state. So if you did a transition in position control mode the commander would temporarily during the transition set the flag_control_position_enabled and flag_control_velocity_enabled to false which made the position controller just control altitude (via throttle setpoint).

RomanBapst, 2018-10-22:
@Stifael @MaEtUgR ping

bresch, 2018-10-26:
Is it maybe related to that part of the code: https://github.com/PX4/Firmware/blob/dc62454f0a0e09b0600885e51017a545a1c668e4/src/lib/FlightTasks/tasks/Auto/FlightTaskAuto.cpp#L515-L534

>// The internal Waypoints might differ from _triplet_prev_wp, _triplet_target and _triplet_next_wp.
	// The cases where it differs:
	// 1. The vehicle already passed the target -> go straight to target
	// 2. The vehicle is more than cruise speed in front of previous waypoint -> go straight to previous waypoint
// 3. The vehicle is more than cruise speed from track -> **go straight to closest point on track**

During back-transition, the vehicle goes faster than the cruise speed. The "internal" waypoint is then set on the track and not on the land waypoint.

RomanBapst, 2018-11-05:
Fixed by #10772

closed 2018-11-06 by RomanBapst