PX4/PX4-Autopilot #10549: Trying to start WIFI connection on TELEM1 does not work

Seeelefant, 2018-09-22:
**Describe the bug**
A clear and concise description of the bug.

**To Reproduce**
I am trying to establish a WIFI conenction on TELEM1 (being miy high power port) port via entering 
`mavlink start -d /dev/ttyS1 -b 921600 -m onboard -r 80000 -x -f`
in the NSH terminal.

However I get no changes
[image]
whereas working via the SYS_COMPANION variable  on TELEM2 everything works.
**Expected behavior**
A clear and concise description of what you expected to happen.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/en/getting_started/flight_reporting.html)).
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- X4
- Pixhawk 2.4.8

**Additional context**
Add any other context about the problem here.

bartslinger, 2018-09-23:
TELEM1 is not necessarily the same as ttyS1. ttyS1 is already in use, as you can see in your mavlink status.

Seeelefant, 2018-09-23:
Hi Bart, thanks for your answer. Is there any method to find this assigment (table etc.) or is the truth only in the sources?

bartslinger, 2018-09-23:
Actually in this case TELEM1 is ttyS1, but that port is already in use by another mavlink instance. You could change this line to make it work for fmuv2/v3 https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rc.mavlink#L56

Seeelefant, 2018-09-23:
Hi Bart,  thanks that helped a lot. I resolved in a slightly different way. I put 
`
\# Stop MAVLink
mavlink stop-all

\# Start MAVLink
mavlink start -r 800000 -d /dev/ttyACM0 -m config -x
mavlink start -d /dev/ttyS1 -b 921600 -r 20000`
`

into /fs/microsd/etc/extras.txt and it worked. Shouldn´t we have an easier possibility to change baudrates in PX4 without the need for a PhD in rocket science? In Ardupilot that is handled by changing parametres.

bkueng, 2018-09-24:
> Shouldn´t we have an easier possibility to change baudrates in PX4 without the need for a PhD in rocket science?

Haha, it's coming: https://github.com/PX4/Firmware/pull/10337.

Closing this then.

closed 2018-09-24 by bkueng