PX4/PX4-Autopilot #10561: Flight Tasks: velocity & attitude setpoints in Acro

bkueng, 2018-09-25:
When flying master in acro I noticed that there are non-zero attitude setpoints in acro mode (but it looks like it only happens when I switched to stabilized and then back to acro). The same happens with velocity setpoints.
Similar in stabilized mode: there are velocity setpoints published.

@MaEtUgR @Stifael is this expected?

Here are 2 logs that illustrate the behavior:
- https://logs.px4.io/plot_app?log=6d419dfb-d12b-49c1-936d-22d750a08870
- https://logs.px4.io/plot_app?log=9d78d9b9-0d19-4d3e-85ce-381e3946f3df

MaEtUgR, 2018-10-30:
I thought the setpoint can only be generated by an active task and is all nan when the task gets initialized which would mean that if the task does not set these setpoints they should be nan. But maybe what is logged is a different state out of the position controller. Let's see.

bkueng, 2018-11-05:
I have a fix for this coming up.

closed 2018-11-13 by bkueng