PX4/PX4-Autopilot #10571: Workaround for a bug in FrSky SBUS failsafe

mijarck, 2018-09-26:
As reported already here the "no pulses" failsafe of FrSky receivers like X8R has a random issue where failsafe flag of the SBUS frame is cleared but data values of some channels show 0 instead of correct channel value
 
https://github.com/ArduPilot/ardupilot/issues/9400

Steps to reproduce the behavior:
- Hopefully by switching off and on the transmitter enough many times and checking the SBUS frames before and after the frames having failsafe flag on

Expected behavior
- SBUS frames having incorrect data dropped

Nurffe, 2018-09-27:
Probably somewhat same issue but other way around can be seen here where failsafe is triggered. Failsafe is set to "no pulses" but still zero is sent for certain channels (5, 9, 10, 11, 12, 14). There is no preset failsafe set in receiver itself so it's unknown why these random channels are set to zero which is then converted to 874 in FC

Here are snapshots from logs

Showing lines and timestamps for radio fs trigger
[image]

Showing lines and timestamps for incorrect channel values
[image]

Error state only lasts ~400ms and happens shortly after failsafe is triggered. Seeing this happen *after* failsafe has been recognized by FC makes me believe there might actually be something in arducopter code itself and not just receiver doing weird things. Ardu shouldn't take nonsense inputs when failsafe flag is already on

dagar, 2018-12-31:
Hi @mijarck,

Which flight controller and firmware version?

mijarck, 2019-02-20:
Was seen with Pixhawk 2.1 and Ardupilot initially. Then repeated on table with receiver only. Didn’t dare to fly too far (where failsafe kicks in) with PX4. Either didn’t tried debugging further about the behaviour on PX4. 

Just for the reference here’s the fix made to ArduPilot master: https://github.com/ArduPilot/ardupilot/pull/10216

closed 2019-07-08 by stale[bot]