PX4/PX4-Autopilot #10575: Trajectory Generator

thomasgubler, 2018-09-26:
Initial investigation and algorithm prototype is done here: https://github.com/bresch/trajectory_generator/tree/master

First discussions on dev call 2018-09-26

bresch, 2018-09-26:
From @MaEtUgR :
> Here is the Smoothing Library located: https://github.com/PX4/Firmware/tree/master/src/lib/FlightTasks/tasks/Utility
It would be probably best if you could add your calculations in a C++ library class in the same way.
Here's the call to the library then for the manual position controlled smoothed flight: https://github.com/PX4/Firmware/blob/master/src/lib/FlightTasks/tasks/ManualPositionSmooth/FlightTaskManualPositionSmooth.cpp#L55-L64

thomasgubler, 2018-10-01:
Status (via @bresch ): first draft/WIP implementation is getting tested in SITL

bresch, 2018-10-02:
Here is the WIP branch:
https://github.com/bkueng/Firmware/tree/trajectory_smoothing
I'm currently developing the [time synchronization](http://www.reflexxes.ws/software/typeiirml/v1.2.6/docs/page__synchronization_behavior.html) needed to provide nice straight lines.

bresch, 2018-10-03:
Time synchronization is done, here is a simulation result on jMAVSim: https://logs.px4.io/plot_app?log=4fc1e3f0-f9ba-48a1-ad78-f2495fc9cbe7

LorenzMeier, 2018-10-03:
Awesome!

closed 2018-11-07 by AuterionWrikeBot