PX4/PX4-Autopilot #10581: EKF2 external vision, yaw divergence issue

Heojongbeom, 2018-09-27:
I use EKF2 estimator with external vision input(MOCAP or msf vision pose).
But sometimes yaw angle of pixhawk(only yaw) is diverged.(0 -> 360 or 360 -> 0)
This is flight log when yaw angle divergence.
https://review.px4.io/plot_app?log=6e7040c7-7a5c-44d8-89b5-80084aa61f33

My system is as follows.
1. Pixhawk firmware 1.8.0 
2. Pixhawk platform is 2.1(cube)
3. EKF2 parameter set
 - EKF2_AID_MASK 24
 - EKF2_HGT_MODE 3 
 - EKF2_MAG_TYPE 4 
 - EKF2_EV_DELAY 0
4. use internal magnetometer(but when I use external magnetometer, divergence is also same)
5. I use MSF with external distance range sensor, 2D lidar, mono camera and IMU of Pixhawk.

Even I have tried replacement pixhawk platform, pixhawk3, pixhawk4, pixhawk2.
But every pixhawk is same result.

The ratio between normal and divergent is almost the same, and in the case of divergence,
it may be okay to turn off the system and turn on and off again.

In the past, I was using the INAV estimator on the pixhawk 1 platform(firmware ver. 1.4.4),
but since pixhawk1 was discontinued, I can not use it anymore. So I have to change it to EKF2

I've been suffering from this problem for almost a month. I ask for your help.

TenVoordeEng, 2018-10-30:
I have a similar problem. I am also using EKF2 estimator with external vision input.

What I see is that sometimes when I startup, the yaw (read from mavros/local_position/pose) is 180 degree rotated compared to what I send (mavros/vision_pose/pose). In that case it is impossible to fly the drone, because it is correcting to the wrong direction.

The strange thing is that the error does not always occur, it is random.

@Heojongbeom can this be the same problem as you described?

Does anyone else see this problem? Any suggestion?

TenVoordeEng, 2018-11-07:
Hi, 

I have disabled the magnetometers. It works without magnetometers. EFK2 does output local pose.

I use this settings (don't know if they are all necessary, but it works)
CAL_MAG0_EN disabled
EFK2_MAG_TYPE None
SYS_HAS_MAG disabled

The vision position and yaw is send correctly to the px4. It takes the position correctly, but the yaw is sometimes 180 degree inverted.

I am using Pixhawk 3Pro with px4 firmware Version 1.8.0dev.

priseborough, 2018-12-03:
The combination of EKF2_MAG_TYPE 4 and EKF2_AID_MASK 24 is untested. I will look at the code for this in more detail this week.

Heojongbeom, 2019-01-02:
Hi

I have tested disabled the magnetometers that you told me, but it still
does not work...
The motion capture data(position and yaw) is send to PX4, but PX4 local
pose does not out...
However
CAL_MAG0_EN enabled
EFK2_MAG_TYPE None
SYS_HAS_MAG disabled
This combination works well, I can't disable all magnetometers

Heojongbeom, 2019-01-14:
Hello Paul

Is any update for these issue?
I really want to use EKF2 + External Vision for our UAV, but we can't until
now because of  yaw angle rotating.
Or, we hope use disabling all magnetometers for using EKF2.
Some guy notified me like this parameters
CAL_MAG0_EN enabled
EFK2_MAG_TYPE None
SYS_HAS_MAG disabled
Do you think it's available parameters? Actually our EKF2 still does not
work in this condition.

Thank you

priseborough, 2019-01-14:
This is not a use case I am familiar with. Your best long term solution is to get a log without significant dropouts from boot and replicate the fault using the ekf2 replay facility using the identical code version you are flying with. Once that can be done, then I will have something that i can work with to understand the root cause and formulate a fix.

closed 2019-07-08 by stale[bot]