PX4/PX4-Autopilot #10584: State Machine Options (Library, Framework, Generator, Compiler, etc)

dagar, 2018-09-27:
I've been casually looking for a good solution to handle state machines throughout the system.
 - https://github.com/PX4/Firmware/issues/7055

**Requirements**
 - declarative
 - generate documentation (either tables or a graph)
     - SCXML would also be nice
 - if a generator or compiler it needs to be easily accessible, cross platform, and have an acceptable license
 - likely need to support HSM (TBD)
 - entry and exit Actions
 - transition guards
 - error handling/reporting (think about feedback from denied arming or mode change)

### Options

 - **State Machine Compiler** - http://smc.sourceforge.net/
      - large java app
      - state machine defined in files then generates source code and documentation
      - doesn't support hierarchical state machines

 - **[Boost].SML (State Machine Language)** - https://github.com/boost-experimental/sml
      - no documentation generation
      - c++14 required
      - don't be afraid of Boost in the name (it's standalone)

 - **ROS Decision Making** - http://wiki.ros.org/decision_making
      - can generate SCXML
      - would need a PX4 port (subset of features)

 - **Microsoft P Language** - https://www.microsoft.com/en-us/research/wp-content/uploads/2016/02/tr-8.pdf
     - can generate C code
     - https://github.com/PX4/Firmware/issues/7055#issuecomment-295638659

dagar, 2018-10-29:
FYI @simonegu @julianoes

mcsauder, 2019-02-15:
@dagar, here is another you can add to your list: https://www.bouml.fr/index.html

dakejahl, 2019-03-04:
@dagar I made a dumb little example in PX4 that uses this library https://github.com/amaiorano/hsm
See https://github.com/PX4/Firmware/tree/hsm_example

It's essentially the `state_args.cpp` with a few modifications. I wouldn't even know where to begin with trying to implement this in something like `Commander`... Hopefully this is somewhat helpful.

`make px4_sitl_default jmavsim`
`pxh > hsm_example`

Right now the `plotHsm.py` doesn't work, but I was able to manually generate the chart:
```
python $PX4_DIR/src/lib/hsm/tools/hsmToDot.py $PX4_DIR/src/modules/hsm_example/hsm_example.h > hsm_example.dot
dot hsm_example.dot -Tpng -o hsm_example.png
```
[image]
**Output**
```
pxh> hsm_example
HSM_1_TestHsm: Init    : Alive
HSM_1_TestHsm:  Entry   : Locomotion
HSM_1_TestHsm:   Entry   : Stand
HSM_2_TestHsm:   Pop     : Stand
HSM_1_TestHsm:   Sibling : Move
HSM_2_TestHsm:   Pop     : Move
HSM_2_TestHsm:  Pop     : Locomotion
HSM_1_TestHsm:  Sibling : Attack
>>> Attacking: 0
>>> Attacking: 0
>>> Attacking: 0
HSM_2_TestHsm:  Pop     : Attack
HSM_1_TestHsm:  Sibling : Attack
>>> Attacking: 1
>>> Attacking: 1
>>> Attacking: 1
HSM_2_TestHsm:  Pop     : Attack
HSM_1_TestHsm:  Sibling : Attack
>>> Attacking: 2
>>> Attacking: 2
>>> Attacking: 2
>>> Attacking: 2
```
I have the debug level set to diagnostic, so you can see all state transitions and their relationships.

dagar, 2019-06-25:
Julian's state machine in PX4/avoidance. https://github.com/PX4/avoidance/pull/418

julianoes, 2019-07-01:
Thanks @dakejahl and @jkflying. That's helpful, however, I still believe the hard part is capturing the intended behaviour and put it into a diagram than to actually implement the state machine. I have implemented a state machine like this in the past as well but was struggling with formulating the actual behaviour without having to do it using hierarchical state machines.

jkflying, 2019-07-01:
Maybe a first step would be drawing a high level state transition diagram and refactoring into that, even if inside the different states there is still a lot of ad-hoc implicit state code?

AuterionWrikeBot, 2019-07-01:
➤ David Sidrane commented:

Have a look at https://www.visual-paradigm.com/ 
It is a great tool and there is a comunity edition

dagar, 2019-07-01:
> I have implemented a state machine like this in the past as well but was struggling with formulating the actual behaviour without having to do it using hierarchical state machines.

Why without hierarchical state machines?

julianoes, 2019-07-02:
Right so for hierarchical state machines you either need to extend the implementations or combine them manually.

dagar, 2019-07-02:
Added https://github.com/andrew-gresyk/HFSM2 to the list to investigate.

wluker, 2019-07-18:
You might consider a language @fsantanna created called ceu. http://ceu-lang.org/index.html

It generates C code that you would have to integrate into an event loop.

It's not a state machine generator, but I find the semantics easier on my brain than state machines, even hierarchical.

Pedro-Roque, 2020-12-21:
Option used by NASA Astrobee: https://github.com/nasa/astrobee/blob/bb0fc3e4110a14929bd4cf35c12b3c169bc6c756/shared/ff_util/include/ff_util/ff_fsm.h

with
https://github.com/nasa/astrobee/blob/bb0fc3e4110a14929bd4cf35c12b3c169bc6c756/gnc/ctl/statemachine.dot

(not generated automatically)

dinomani, 2021-01-06:
@dagar 
Anybotic uses something like that to switch the robot to and from different modes with safety fallback.